betaflight/betaflight #1732: gyro data recording not readable

TheAngularity, 2016-12-02:
today i make a test flight ... no special happenings noticed ... super fluid flight ... copied blackbox data 
and now whats wrong??? ... cant not processing the gyro data ... something is wrong with the gyro data recording ... here is a screenshot and the blackbox file

[image]

[LOG00008.TXT](https://github.com/betaflight/betaflight/files/627211/LOG00008.TXT)

martinbudden, 2016-12-02:
PR https://github.com/betaflight/betaflight/pull/1715 - Moved sensor global data into sensor_s structs also changed the way blackbox data was written. Now the data written to blackbox is in degrees per second, when previously it was raw data.

There are two possible fixes:
1. Changed betaflight back to writing raw data
2. Change the blackbox view to use degrees per second.

I favour the second. @borisbstyle what do you think?

borisbstyle, 2016-12-02:
@GaryKeeble 
What do you think about this? Should we change the logging units in the blackbox?

ledvinap, 2016-12-02:
There is already `gyro_scale` in blackbox header, writing correct value for DPS should be enough ..

martinbudden, 2016-12-02:
Yes, I thought I might just have set the scale wrong.

ledvinap, 2016-12-02:
The header posted by @TheAngularity has `H gyro_scale:0x3d79c190`. That is 1/16.4 (0.0610), but now the scale should be 1.0 (the value is already scaled).
The strange this is that 16x smaller gyro reading shall not be that big problem (IMU simulation won't work .. )

ledvinap, 2016-12-02:
Using blackbox_decode, the data look reasonable ...

martinbudden, 2016-12-02:
@ledvinap, IMU code takes into account new scale

ledvinap, 2016-12-02:
Blackbox decoder/viewer? It does simulate IMU to get orientation .. (but TheAngularity data do not contain ACC)

GaryKeeble, 2016-12-02:
There's a couple of issues here,
1. It would be better to put gyro scale to 1.0 in the flight controller so that the viewer could still be used to view older logs without change; but we could change the viewer and add a version check if people prefer.
1. The header field used to be called "gyro.scale" and has been re-labelled "gyro_scale"; this has broken the viewer as it no longer reads the scale; again, we can change the log viewer and accept either "gyro.scale" or "gyro_scale"; which is probably better in this case as "gyro.scale" is not consistent with other header entries;

I modified the log file above changing the header line to **H gyro.scale:0x3f800000** and I beleive the log viewer displays correctly.

GaryKeeble, 2016-12-02:
With respect to VBAT, has anything changed with "vbatReference" it seems to be a factor of 10 larger?

TheAngularity, 2016-12-02:
> It would be better to put gyro scale to 1.0 in the flight controller so that the viewer could still be used to view older logs without change; but we could change the viewer and add a version check if people prefer.

I think that is the simplest solution ... I would prefer this

> The header field used to be called "gyro.scale" and has been re-labelled "gyro_scale"; this has broken the viewer as it no longer reads the scale; again, we can change the log viewer and accept [...]

edited the `flightlog_parser.js` and the logfile ... then again everything is ok

GaryKeeble, 2016-12-02:
Modified log viewer master to handle gyro_scale and default value to 1.0. 
betaflight/blackbox-log-viewer#38

TheAngularity, 2016-12-02:
posted my response in [betaflight/blackbox-log-viewer#38](https://github.com/betaflight/blackbox-log-viewer/issues/38) ... now i think i can close this

martinbudden, 2016-12-02:
Reopening because there are some changes to be made in betaflight.

TheAngularity, 2016-12-02:
@martinbudden 
ok ... I have closed to avoid overlapped response

closed 2016-12-19 by borisbstyle