betaflight/betaflight #1734: make gyro notch settings profile dependent?

andre92, 2016-12-02:
I use different pid profiles for different props and I noticed I need different gyro notch settings for each prop to work well (noisy 450 copter) My FC is a SPRacing F3 EVO with mpu6500 so I need gyro notch filter.
since Dterm filtering is already in profile, is it possible to move all filter settings into profile?

borisbstyle, 2016-12-05:
This is not possible as gyro is first thing loaded during boot even before profile stuff like PID's are loaded.

ledvinap, 2016-12-05:
@borisbstyle : with profile switching, some way to apply different filter parameters after profile change is necessary. It will solve boot order problem as side effect ...

andre92, 2016-12-05:
update:  I mean only the soft filters. I do realize it is not possible to change the mpu6500 filter (gyro_lpf) runtime, but as profiles are changed, maybe it is possible to call the init function for biquad filters?

borisbstyle, 2016-12-06:
@ledvinap @andre92 
Technically this is already possible for soft filters. One of the last cleanups from @martinbudden already has gyroInit() for filter reinitialisation and can be done on the fly through MSP.

I am only not sure if we want more profile parameters? The current code base offers changes through OSD and smartport as well so one should be able to change it on the fly.

ledvinap, 2016-12-06:
@borisbstyle : Andres use case sounds reasonable. Also filter may be fine-tuned for PID paratemers / flystyle

hydra, 2019-07-14:
is this still required?

mikeller, 2019-07-14:
Overhauling what types of profiles there are, and what is / isn't part of what type of profile is one of the cleanup jobs that should be done at some point. Not sure if we want to keep this, or close it as we are going to start with a green field approach for assigning parameters to profiles.

closed 2022-05-09 by haslinghuis