betaflight/betaflight #1735: Proposal to add averageSystemLoadPercent to MSP_STATUS

SandroGrzicic, 2016-12-03:
Since I wanted to add support for showing additional data via a separate OSD, I was looking into what was being sent using MSP_STATUS:

Both CF and BF send cycleTime, I2C error count, a bitmask of enabled sensors, a bitmask of flight modes, and the current profile.

Both CF and BF additionally send the CPU load (`averageSystemLoadPercent`), however CF sends it immediately after the profile in MSP_STATUS, while BF has a completely separate MSP command called MSP_STATUS_EX in which the CPU load is sent immediately after the profile:

CF: https://github.com/cleanflight/cleanflight/blob/master/src/main/fc/msp_server_fc.c#L547

BF: https://github.com/betaflight/betaflight/blob/master/src/main/fc/fc_msp.c#L575

Additionally, CF also sends [`gyroDeltaUs`](https://github.com/cleanflight/cleanflight/blob/master/src/main/fc/cleanflight_fc.c#L807), while BF also sends `MAX_PROFILE_COUNT`, as well as `getCurrentControlRateProfile()`.

As far as I know (please correct me here if I'm wrong), adding new fields to an existing MSP cannot result in compatibility issues.

So why weren't these extra fields simply added to MSP_STATUS in BF?

Namely, since I don't see a reason against BF/CF compatibility in such a basic MSP, I suggest I create a PR with the following change - add the `sbufWriteU16(dst, averageSystemLoadPercent);` at the end of MSP_STATUS.

_Optionally, I suggest I also copy the `gyroDeltaUs` field from CF (can be set to zero in BF if it has no meaning - I have no idea what this is), and also add, `MAX_PROFILE_COUNT` and `getCurrentControlRateProfile()` fields from MSP_STATUS_EX, in both CF and BF, so in the end the MSP_STATUS response would have an identical format across CF and BF._

Thoughts? I am completely new to CF/BF so if this is completely infeasible, instead of "fixing" MSP_STATUS, it's of course possible to skip changing MSP_STATUS and do my original task "the hard way" by detecting CF/BF in the MSP client and then request either MSP_STATUS or MSP_STATUS_EX, as appropriate, but I feel like this is unnecessary and against good programming practice as the MSPs are semantically identical. Again, I don't understand why was the _EX needed in the first place.

Related: https://github.com/betaflight/betaflight/pull/866

Thanks!

nathantsoi, 2016-12-03:
Good idea. The problem is that any change to an existing MSP command requires version gating (like this https://github.com/betaflight/betaflight-configurator/commit/2fb9a87906db2a4aa4f5080486101f44fd81a9e6) and testing.

It would be really cool if we had a protocol that could serial/deserialize itself... we talked about implementing [nanopb](https://koti.kapsi.fi/jpa/nanopb/) or similar. It would be a cool thing to explore if you're interested in prototyping it out.

SandroGrzicic, 2016-12-03:
Thanks for the feedback.

As I said, wouldn't adding new fields, without modifying existing ones not break anything? Or am I misunderstanding how the deserialization works? Is the buffer after reading finished destroyed or does it still contain the remaining data, corrupting the next read? I am looking at `mspSerialProcessReceivedCommand`, I think I need to read it a bit more to understand what's exactly going on.

PB would be great, I worked with Protocol Buffers and even wrote a Scala compiler for it, ScalaBuff. However, I think that switching to another serialization protocol would completely destroy compatibility with existing MSP clients, right? Unless of course the old serialization code is left in, but that would be a maintenance nightmare..

mikeller, 2016-12-06:
@SandroGrzicic: Fair call, I think it would make sense to add the fields that are missing in `MSP_STATUS` in Betaflight compared to cleanflight. This will not break anything, and it will make life easier for consumers of MSP messages.

The reasond for these fields to be sent in `MSP_STATUS_EX`  in Betaflight is that they were added to Betaflight first, and at the time the implementer wanted to avoid breaking compatibility with the `MSP_STATUS` command that was implemented by cleanflight.

PR #1762 (2016-12-08): Make 'MSP_STATUS' compatible to implementation in cleanflight.
Fixes #1735.
closed 2016-12-08 by mikeller