betaflight/betaflight #1736: 3.1 Revo F4 Target Buzzer Constant beep

puaka, 2016-12-03:
After flashed, buzzer constant beeping. VBAT not connected and disabled under configuration. ESC not yet flashed with DSHOT. No problem on 3.0.1.

AndersHoglund, 2016-12-03:
What FC? Where/how is buzzer connected? Buzzer type, buffered or unbuffered? If buffered, is it PnP or NpN ? Any attempts done to fix it by using resource mapping?

My RTFQ Revo Acro works just fine, NpN buffered beeper on motor output 6 using these settings:

    # version
    # Betaflight/REVO 3.1.0 Nov 23 2016 / 15:41:51 (4040a1d)

    # resources
    resource BEEPER A00
    resource MOTOR 6 NONE

puaka, 2016-12-04:
Its from banggood. Revo F4. It has buzzer pad, it works fine on 3.0.1 but not on 3.1 745

AndersHoglund, 2016-12-04:
Well, this FC is not exactly a REVO. Its more like an AirbotF4. This FC has it's own firmware from v3.1

puaka, 2016-12-04:
Ahhh that make senses. Will try the AirBotF4. They do look indentical :) Thanks

puaka, 2016-12-05:
Awesome! It is fine now, everything works as expected. So revo F4 from banggod which has buzzer pad need to use AIRBOTF4 target.

closed 2016-12-05 by puaka