betaflight/betaflight #1742: [SPR F3 Evo] Blackbox/SD card not working on SP Racing F3 Evo with sdcard_dma=on

SandroGrzicic, 2016-12-03:
I've just spent an hour trying to make my SD card reader on my SP Racing F3 Evo work with the [latest BF master](5ee5ec135867fb5a70f8bd20be6460079db2e4db). After trying everything, including reformatting the (Turnigy 32GB SDHC) card several times. I finally made it work by randomly stumbling upon a configurator setting called `sdcard_dma`. It was enabled. Presumably due to all the recent DMA changes, this left the SD card reader DMA broken in some way. 

After setting the variable to 'off', the SD card reader now works fine and the resulting .BFL logs can be opened in the Blackbox Viewer.

Marax, 2016-12-06:
There is bug also reported for EVO with 3.0 https://github.com/betaflight/betaflight/issues/611

Ede2016, 2016-12-18:
I just tested this with the SPro Mini and I confirm it has the same problem like the EVO described above.
I use DShot 150 and I used: resource PPM NONE, resource MOTOR 4 B04

I tested it with release 783 and with 831
If sdcard-dma=on then log files with the size 4MB are created but they can't be read. If I try to read them I see the error: Log is truncated, contains no data

If sdcard-dma=off then the log files have different sizes and can be read.

With the blackbox explorer 2.5.7 at least some data are shown incorrect
With the blackbox explorer 2.5.8 all looks fine (at least on the first view)

Sometimes when I armed the quad and sdcard-dma=on motor 3 did not spin. With sdcard-dma=off I did not observe this problem. But I don't know if this is related to above. I tested it just a few times.

My knowledge is limited but I am willing to do more tests it required. Please let me know if and what I should test.

SandroGrzicic, 2016-12-18:
It looks like a DMA conflict between the SDCard reader and the motor3 output in your case.

These are presumably the relevant #defines for your FC:

target.c
````
DEF_TIM(TIM4,  CH3, PB8,  TIM_USE_MOTOR, 1),  // PWM3 - PB8
````

target.h
````
#define USE_SPI
#define USE_SPI_DEVICE_2 // PB12,13,14,15 on AF5

#define SPI2_NSS_PIN            PB12
#define SPI2_SCK_PIN            PB13
#define SPI2_MISO_PIN           PB14
#define SPI2_MOSI_PIN           PB15

#define USE_SDCARD
#define USE_SDCARD_SPI2

#define SDCARD_DETECT_INVERTED

#define SDCARD_DETECT_PIN                   PC14
#define SDCARD_SPI_INSTANCE                 SPI2
#define SDCARD_SPI_CS_PIN                   SPI2_NSS_PIN

// SPI2 is on the APB1 bus whose clock runs at 36MHz. Divide to under 400kHz for init:
#define SDCARD_SPI_INITIALIZATION_CLOCK_DIVIDER 128
// Divide to under 25MHz for normal operation:
#define SDCARD_SPI_FULL_SPEED_CLOCK_DIVIDER     2

// Note, this is the same DMA channel as UART1_RX. Luckily we don't use DMA for USART Rx.
#define SDCARD_DMA_CHANNEL_TX               DMA1_Channel5
#define SDCARD_DMA_CHANNEL_TX_COMPLETE_FLAG DMA1_FLAG_TC5
````

However, I don't know what to look for exactly here to find what the issue is.

Ede2016, 2016-12-18:
@hydra Dominic Clifton, maybe you can have a look at this. You are the developer of the EVO and the MINI and I am sure you are the most qualified person to address and hopefully solve these problems.

borisbstyle, 2016-12-19:
I will mark this as hardware limitation. I believe Spf3 EVO does not have possibility to run dshot on existing motors. 
Try resource command to move one of the motors to other pin.

@blckmn You tested Evo before right?

SandroGrzicic, 2016-12-19:
Note that my issue is about the SDCard, with DShot off. Do you think that even without DShot enabled, the recent changes cause a DMA conflict with SDCard?

borisbstyle, 2016-12-19:
Can you try again on the latest master.

Ede2016, 2016-12-20:
For clarification. My description above is about the SPro MINI and not the EVO. I added it to this issue because it is the same or at least a similar problem.
The MINI works without problems without DShot.
DShot only works for all motors if motor 4 is remapped to the PPM pin. This is now described in the Dshot & BetaFlight  Wiki.
On the MINI the problem with the SDCard as described above happens only if DShot is enabled.
I leave it up to the professionals if you want to separate this issue into to issues separated for the EVO and the MINI.

hydra, 2016-12-20:
@Ede2016 If you're getting issues even with DSHOT off then it sounds like a regression bug unrelated to DSHOT which should be fixed when someone has a chance.

borisbstyle, 2016-12-20:
@Ede2016 
Can you post the "resource" dump? Why so confusiont issue? Is it now Mini or EVO?

Ede2016, 2016-12-20:
@borisbstyle the original issue is about the EVO. I added my comment above because I have a very similar issue with my MINI. I don't have an EVO and cannot comment about it.
Sorry, I won't be able to do the "resource" dump within at least the next 12h or so.

hydra, 2016-12-20:
@Ede2016 if you have an issue with the mini then open a new issue with full details of your issue and reference this one from the comments.  It's hard to get things done and make sense of the comments when the conversation goes off the original topic.

borisbstyle, 2016-12-20:
yes please open a new issue and tested on the latest master. This is confusing when multiple issues are discussed under same issue

closed 2016-12-20 by borisbstyle

Marax, 2016-12-20:
Finally got time to test it on my SPF3EVO with latest build (836).
sdcard_dma = **on** blackbox doesn't work - after filesystem initializing I get error
sdcard_dma = **off** blackbox works

SandroGrzicic, 2017-01-02:
It looks like the issue is solved - the default setting of `sdcard_dma on` seems to result in a working SDCard on the Evo FC, however I haven't tested this in combanation with DShot.

Ede2016, 2017-01-02:
@SandroGrzicic  The problem exists only with Dshot enabled. There never was any problem with  sdcard_dma=on for OneShot or any other of the "traditionally" protocols.

Marax, 2017-01-30:
Sorry guys, but issue is still present on BF 3.1. Using SPF3EVO with DSHOT. Sd card contains FREESPACE.E of 0kb and sd_info reports Startup failed.
@borisbstyle is there something I could try to help resolve this? Or is this some kind of HW limitation of EVO board?

raphaelcoeffic, 2017-01-30:
@Ede2016 From what I tested today, this is related to UART2. As soon as I enabled serial RX on UART2, the issue starts showing up. Otherwise, it just works. Please note that this absolutely not dependent on whether or not DSHOT is used. In my case, setting up SBUS on UART2 is the only configuration item differing from the defaults.

Marax, 2017-01-30:
@raphaelcoeffic I'm also using UART2.

emski0, 2017-02-04:
@Marax did you set sdcard_dma = off,??  you need to set it to off, and then the sdcard logging will work .
I am also using sbus on UART2  with dshot 600, no problem here,everything is working

raphaelcoeffic, 2017-02-04:
@emski0 the original issue (and mine as well), is that it does not work either WITHOUT DSHOT.

hydra, 2017-02-05:
Removed the 'hardware limitation' tag since there is no hardware limitation that prevents the use of sd card with dma writes.  There maybe a limitation WITH dshot, but since @raphaelcoeffic noted this issues is unrelated to dshot and is just a normal bug which should be fixed.

suggest doing a git bisect from when it DID work to find out when it stopped working so the change that broke it can be identified.