betaflight/betaflight #1749: hard roll at min throttle leads to uncontrolled roll

scottpetrlik, 2016-12-05:
Sometimes, when I do a hard roll, the quad looses control and spins wildly.

I've destroyed two different quads this way, both running betaflight 3.0.

In both cases, BlHeli 20amp esc's, running oneshot 125
One quad ran a lumenier lux, the other ran a Seriously Dodo FC.

at 1:01-
[LOG00021.TXT](https://github.com/betaflight/betaflight/files/630021/LOG00021.TXT)

at 1:00-
[blackbox_log_2016-12-04_202110.TXT](https://github.com/betaflight/betaflight/files/630012/blackbox_log_2016-12-04_202110.TXT)

lostcontrol, 2016-12-05:
Most likely not a software bug so you should better seek help on rcgroups.com. Did you try increasing min_throttle?

scottpetrlik, 2016-12-05:
How does min_throttle affect the problem?

What is the best way to set min throttle? I set it to the min value that made my motors spin.

-Scott

borisbstyle, 2016-12-05:
Sounds like a typical ESC desync

rav-rav, 2016-12-05:
> What is the best way to set min throttle? I set it to the min value that made my motors spin.

This is most likely the issue. Your setting might be enough to make them spin on ground but not in the air with wind etc. The motor will desync and your quad will spin wildly.
Add at least 20, better 30 or 40 to the value where your motor starts to spin to be sure that it will not stop midair.

borisbstyle, 2016-12-19:
closing this one as it is known issue on blheli escs

closed 2016-12-19 by borisbstyle