betaflight/betaflight #1755: I2C error handling initializes bogus I2C bus

jflyper, 2016-12-05:
Happens when
- `USE_I2C` is defined
- `I2C_DEVICE` is defined as other than `I2CDEV_1` (say `I2C_DEV2` hereafter)
- Feature `DASHBOARD` (`DISPLAY`) is turned on

Happens because
- `main.c` calls `i2cInit(I2C_DEVICE)` and initializes `I2C_DEV2`.
- OLED driver (`drivers/display_ug2864hsweg01.c`) looks for `OLED_I2C_INSTANCE`, which is set to `I2CDEV_1` when not defined.
- DASHBOARD tries to write to `OLED_I2C_INSTANCE`(= `I2C_DEV1`)
- i2c driver detects an error on `I2C_DEV1`, and calls `i2cInit(I2C_DEV1)` as a part of error handling (`i2cHandleHardwareFailure`).
- The call to `i2cInit(I2C_DEV1)` initializes `I2C_DEV1`.

Possible fix
- Make the `OLED_I2C_INSTANCE` configurable.
- Make the OLED driver scan for initialized I2C bus.
- Make `i2cHandleHardwareFailure` to call a new `i2cReInit`, which does everything that `i2cInit` does except for resource allocation. The `i2cReinit` rejects non-initialized instance.
- Others?

borisbstyle, 2016-12-20:
Is this related to
https://github.com/betaflight/betaflight/issues/1582

jflyper, 2016-12-20:
@borisbstyle 
It's a different issue, but yes, it's related.

When targets start to use I2C other than I2CDEV_1 (I2C1), this issue bites.

closed 2017-07-24 by jflyper