betaflight/betaflight #1756: SPRacingF3 Target Cannot Flash ESC with Pass through

Norcal3d, 2016-12-05:
A SPRacingF3 Target (MRM Mantis F3) cannot flash ESC after updating to 3.1 (tried up to build 789. You can Update settings, but flashing firmware will not work. Works as expected on 3.0.1.

DieHertz, 2016-12-07:
Please, provide logs from your Flashing utility.

Norcal3d, 2016-12-07:
[Errors.zip](https://github.com/betaflight/betaflight/files/637114/Errors.zip)

Here is a log trying to flash a file from the affected board using BLheli windows application, not chrome configurator

Norcal3d, 2016-12-07:
[Log-ok.zip](https://github.com/betaflight/betaflight/files/637128/Log-ok.zip)

Here is a log of flashing in the same manner, successfully, after reverting to BF3.0.1.

DieHertz, 2016-12-07:
Thanks for you help, @Norcal3d!
May I ask you for another favor?
Reproduce a failed flash operation, then go back to BF3.0.1 and make complete dump of ESC memory, attach it here.
You can make a dump on the `Expert Tools` page, set:
```
Start = 0x00
Length = 0x2000
```
Save resulting file either as `HEX` or `BIN`.
That should be enough for us to investigate the issue

Norcal3d, 2016-12-07:
No, thank you! Ill get that dump tonite. I don't have a solder iron with me in case of needing a bootloader jump when flashing.

DieHertz, 2016-12-08:
Your ESCs are DYS XS20A? Have you performed any modifications on them like removing caps for DSHOT?

Norcal3d, 2016-12-09:
Sorry for the delay. Tested on my Dodo Rev3 with same results. Yes DYS XS20 & FVT littlebee 20 both with CAP removed. Here is teh Memory dump after a bad flash on 3.1.

DieHertz, 2016-12-09:
Seems like file attachment got lost :-(

Norcal3d, 2016-12-09:
here you go.

[DYS XS20 - SPracingF3 Bad on 3.1.zip](https://github.com/betaflight/betaflight/files/642443/DYS.XS20.-.SPracingF3.Bad.on.3.1.zip)

DieHertz, 2016-12-09:
Thank you! That should be enough.
Judging by the flash dump, flash erase & write operations succeeded, memory contents from address 0x400 up to 0x1A00 are completely identical to the reference HEX file for A-H-70.
What fails is the first read operation after that, failing to verify previous writes. Luckily the flashing algorithm is written to ensure bootloader operation on such failed attempts.

Why FC -> PC link breaks is a different question, we'll have to debug and study it on BF3.1, apparently it did not happen previously.
Hopefully me or other BF dev will be able to reproduce it under SWD.

DieHertz, 2016-12-09:
I have reproduced the issue locally. I'll look into it soon.

DieHertz, 2016-12-10:
`drivers/serial_uart.c` being optimized at `Ofast` instead of `Os` is the "culprit". Will study the code to figure out what actually breaks, maybe compare resulting binaries.
There's likely some memory error which is not revealed unless using higher optimization levels.

DieHertz, 2016-12-10:
Actually the latest builds do not exhibit this bug, commit 6ea054dcf3c8c6a4dd77472c0d730a166dd8ee45 fixes it.

AndersHoglund, 2016-12-10:
Hmmm, "... fixes it". You probably mean ".... hides it".

DieHertz, 2016-12-10:
@AndersHoglund Yes, my previous sentence has a hint that it is just hidden. :-)
That's why I'm studying code now to find non-compliant parts.

borisbstyle, 2016-12-10:
Escserial also needs to be added to non Rt list

DieHertz, 2016-12-10:
Last time I checked it was there.

Norcal3d, 2016-12-10:
yes the _fix_ has helped. Thanks for all you guys do!

DieHertz, 2016-12-10:
@Norcal3d You're welcome, thanks for cooperating!

ledvinap, 2016-12-11:
It is  sometimes possible to use 'pragma optimize'  to bissect the problem

PR #1815 (2016-12-12): Prevent serial buffer overflow in BLHeli Passthrough code
This PR fixes #1756 and #1728.
closed 2016-12-12 by borisbstyle