betaflight/betaflight #1758: BF 3.1 build 782 acc drift

Jahnkeanater, 2016-12-05:
I have had this issue in the past with certain releases and it seems like its back. In horizon mode the accelerometer loses calibration after doing rolls. To maintain level requires a large amount of correction in the opposite direction. Resetting the FC fixes the issue but it comes back right after doing a roll.

Also the defaults make horizon and angle flight modes incredibly sensitive. Maybe change the default max angle from 70.0 to something more reasonable like 45.0 Or possibly add expo to autolevel that is independent from the acro flight mode.

acro mode works flawlessly :)

borisbstyle, 2016-12-06:
you can change the level_sensitivity by yourself in the cli. 

Acc drift is usually caused by noise. Try soft mounting perhaps

Jahnkeanater, 2016-12-07:
Its not really a drift as much as after a flip it locks onto a different angle. The FC is pretty well isolated from vibrations. Silicone tubing and foam servo tape. Maybe I just need to tune it better?

Ill have to play around with level_sensitivity. Its all the way at the bottom of the cli. :)

andre92, 2016-12-07:
@Jahnkeanater it is probably because of the IMU update rate and gyro vibrations. Acc reacts too slow to upset the IMU in just one flip. IMU also relies on the gyro for integration and if the gyro isn't clean, it may show aliasing drift at 100hz. Maybe increasing the rate for task_attitude to 200-500hz helps reduce gyro aliasing in the IMU.
Default max_angle of 700 is just fine, it is just the very sensitive stick feel of 2.00. I feel a default of 1.00 would be better.

commit 25c7db49 (2016-02-04): Allow PASSTHRU mode when using custom airplane mixer. fixes #1758
closed 2016-12-19 by borisbstyle