betaflight/betaflight #1767: revolt f4 motor mixer issue

JUSTBYCHANCE, 2016-12-06:
i flashed a revolt f4 with 3.1 everything works fine until takeoff.  I have everything on correctly(props) and motors spin right way.  Using mixer quad x 1234 and wired esc's as setup on revolt.  Quad freaks out upon lift.  Changed wiring to esc's to match that of quad x, same thing.  arm up great and motors spin fine, freak out and crash upon takeoff.  Again, triple checked props and motor rotation, and its good.  Ideas?

borisbstyle, 2016-12-06:
Test through motor tab each motor and see if it properly responds

JUSTBYCHANCE, 2016-12-07:
They do when wired from original outputs of to the motors but those locations in accordance with the board layout of the revolt don't lie at the quad x motor layout.  As in motor one pin is top left but motor one is rear right.  But it still goes crazy!

andre92, 2016-12-07:
did you check board orientation as well?

JUSTBYCHANCE, 2016-12-07:
Yes

blckmn, 2016-12-08:
The motor numbers marked on the board aren't aligned. If you wire up based on motor output closest to ESC or frame Arm (as in no wires cross crossing the board) then you can use quadx.

You can also play around with the motor tab and move the motor outputs using the resource command.

ZenDumbo, 2016-12-11:
I have posted it in betaflight Facebook group but here are what we need to do.

If people using revolt FC with the raceflight pinout diagram, can do the following for BF firmware:

resource MOTOR 1 A03
resource MOTOR 2 B01
resource MOTOR 3 A02
resource MOTOR 4 B00

Also the board rotated 90 degree yaw with BF:
align_board_yaw = 90

blckmn, 2016-12-11:
Thanks @zendumbo. We've just merged those you've provided as the defaults - so I'll close this issue. Builds from today should be correct out of the box. Thx.

closed 2016-12-11 by blckmn

raamfp, 2017-01-10:
just flashed 3.1 rc6 on my revolt. i tried all settings above and didn't work for me. when i change to Quad X 1234 the diagram doesn't change so motors still 4231. any idea? my board is wired as race flight revolt suppose to be motor starting from 1234. thanks

mikeller, 2017-01-11:
@raamfp: Have you actually tested if the motors are changed or not? I suspect it might just be a case of the diagram being wrong.

raamfp, 2017-01-11:
Yes sure i did check also on motor tab and still run's 4231.. also tried to takeoff but it flipped like crazy. Board orientation is ok. Weird is that the option QuadX 1234 is there but has no effect. 3.1 flies sooo good on my F3 i can imagine what it can do with the revolt F4, just dont want to resolder everything... but anyway amazing work everyone is doing thanks a lot!!!!

mikeller, 2017-01-11:
Did you use the REVO target or the (new) REVOLT target?

raamfp, 2017-01-11:
Revolt target

mikeller, 2017-01-11:
This is weird. The REVOLT target uses exactly the resource assignments that you pointed out as working above:

https://github.com/betaflight/betaflight/blob/master/src/main/target/REVO/target.c#L35-L39

Can you please try resetting your flight controller to defaults, to make sure resource assignments are reset?

AndersHoglund, 2017-01-11:
Mount motors and connect them according to the RF order, i.e. M1 front left, M2 front right, M3 rear right and M4 rear left. In BF there are already a re-mapping by default, as of above, so BF motor 1 is RF M3 etc. 
You should NOT do any resource re-mapping or physical re-arrangement of motors. It is already taken care of for you.

raamfp, 2017-01-11:
Thats what i did its all done following raceflight order...without remapping or remmaping i get same results and still motors spin 4231... and order RF M1 is BF M4 not 3... right?

raamfp, 2017-01-11:
Mikeler i also tried erase board reflashed raceflight, re flashed betaflight and problem persists... just dont want resolder everything again :)

blckmn, 2017-01-11:
What mixer have you chosen? Which motor spins for motor 1 in the motors tab, not the board markers, but position FR, FL, RR, or RL?

raamfp, 2017-01-11:
Mix 1234 like raceflight... and still in the motor tab when i spin motor 1 will spin like on betaflight RR 4231 ...

raamfp, 2017-01-11:
I mean as i said i tried everything that is on here... folowed every single option here... used the regular betaflight mix 4231 not remaping channels or remaping, also mix 1234 and motors still run 4231.

AndersHoglund, 2017-01-11:
Use the default QuadX mixer.
If Motor 1 in Motor Test spins the RR motor (RF M3), then everything is OK. Why do you think there is a problem?
Reset to defaults. Do not change anything. Motor mapping is already taken care of.

blckmn, 2017-01-11:
Yes - leave it on standard QuadX mixer. No need to change it.

raamfp, 2017-01-11:
I tried that too guys.... same problem.

Eganwp, 2017-01-11:
You are ABSOLUTELY positive that your motor signal wires are soldered on the fc in the correct motor and that you didn't use a wrong wiring diagram or a mis-silkscreened board?

raamfp, 2017-01-11:
Ok guys ill go again... my revolt is fine works perfect on raceflight! Its soldered following raceflight diagram 1234. I flashed betaflight and no matter wich option i use motors still run like betaflight! I just reflahsed again and changed nothing and still wrong!

raamfp, 2017-01-11:
I just want to use betaflight on my revolt without having to re solder everything again... raceflight latout is much better with the pads close to the motors.

wind0r, 2017-01-11:
You do not need to resolder! You can simply switch the pins around with the resource command.

also why is it wrong? Just because it isnt the same as raceflight?
Just test Pitch/Yaw without props to check if all is fine. 
That the labels on the Board to not align with labels in the configurator isnt a problem

raamfp, 2017-01-11:
Mate there is nothing wrong....its alll goooooood!!!!! Just trying to solve my puzzle here!!!! i tried remaping too... no go. I mean there is a option on betaflight for QuadX1234 but doesnt work...

martinbudden, 2017-01-11:
@wind0r , @raamfp has a legitimate problem. The resource command shouldn't be required since the pin mapping is handled in `target.c`

@raamfp , can you provide the output of the `diff` command?

martinbudden, 2017-01-11:
@mikeller , @AndersHoglund , it's possible that `GYRO_MPU6500_ALIGN` and `ACC_MPU6500_ALIGN` is incorrectly defined in `target.h`

wind0r, 2017-01-11:
@martinbudden yes it is a legitmate problem resource command should not be needed.
doing a remap with resource was just a quick fix suggestion.
since I do not have a revolt i asked @blckmn to recheck the revolt target.

from other people around i know that the target was working correctly, so i just want to ask for a better description of the problem. since maybe at the end it is just an interface error
since do not work isnt any help for me or other people around here.

personaly i think the mixer is correct but the alignment was wrong and that is why it flipped

AndersHoglund, 2017-01-11:
I don't get this. Do you have a problem or not ?? 
You say both "there is nothing wrong" and ".its alll goooooood" and then "trying to solve my puzzle". What puzzle ? 
There is no need to resolder, re-arrange motors, re-mapping resources, changing mixer etc etc. Keep everythinh is it was when flying with RF. 
There is no puzzle to solve. By thinking there is and trying to solve that, you are actually creating a problem.

raamfp, 2017-01-11:
Hey martin i have no idea how to output the diff command... but if there is a QuadX1234 diagram that should work right? ... i did left acelerometer on and copter orientation as well the gyro is fine!

raamfp, 2017-01-11:
Ahahha mann this is getting funny! Yes there is a problem a option that is not working... i flash my revolt with Betaflight and cant change the motors diagram to follow the raceflight diagram.

raamfp, 2017-01-11:
If its too much trouble i just fly raceflight... im just trying to point a problem to be fixed.

raamfp, 2017-01-11:
Board alignment is perfectly fine no need to yaw 90...

wind0r, 2017-01-11:
as @blckmn pointed out at 11 Dec 2016 he fixed the alignment.
Maybe he forget the change the motor alignment.
Lets all calm down and wait for a response from @blckmn.
Since he have a revolt target to test the changes it will be a realy easy at short task for him :)

Lets all have a good day :)

martinbudden, 2017-01-11:
@raamfp , in the configurator go to the tab marked `CLI`. Then type `diff`. Then copy the output and paste in this thread. That will give us a basis for analysing the problem.

raamfp, 2017-01-11:
$M>lõÿf
Entering CLI Mode, type 'exit' to return, or 'help'

# diff

# version
# Betaflight / REVOLT 3.1.0 Jan 11 2017 / 09:02:10 (322ae59)

# name

# resources

# mixer

# servo

# servo mix

# feature
feature -RX_SERIAL

# beeper

# map

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# rxfail

# master

# profile
profile 0

rateprofile 0

# rateprofile
rateprofile 0

raamfp, 2017-01-11:
ok so on motor tab... Spin motor 1 will spin back right, motor 2 front right, motor 3 back left, motor 4 front left. If i change to QuadX1234 will spin the same.

wind0r, 2017-01-11:
so all is fine :)

AndersHoglund, 2017-01-11:
OK, that is the correct motor order. No problem there.

raamfp, 2017-01-11:
no guys, if u try to take of will flip.

raamfp, 2017-01-11:
when i spin motor one it should spin front left, 2 front right, 3 back left and 4 back right.

AndersHoglund, 2017-01-11:
Nonono. that is RF. You are in BF now.

AndersHoglund, 2017-01-11:
Look at the image in BFC and the motor numbers there.

wind0r, 2017-01-11:
Just look at the Picture in the Configurator. 
as i said before the numbers on the Board are not important.

Just check without props if the quad yaws and pitch correctly.
The flip you got some Weeks ago was because of the not correct setup gyro, which already got fixed.

raamfp, 2017-01-11:
i know beta flight is different.... I'm not a complete noob on this :D ... but esc's are soldered following race flight diagram i understand that you guys are pointing that i have to use betaflight diagram now and i tried that too and still flips!!

martinbudden, 2017-01-11:
OK, some more tests. All WITH PROPS OFF

1. Arm quad. Give a small amount of throttle. Push pitch stick forward. Note which motors go faster and which go slower.
2. Do the same, except push the roll stick right. Again note which motors go faster and which go slower.
3. Place quad on a smooth flat surface. Give it a medium amount of throttle. Push the yaw stick right. Which way does the quad spin?

raamfp, 2017-01-11:
reflashed RC7 and it worked like you pointed Martin no changes needed. not flipping anymore. Maybe it was a RC6 problem. thanks for the patience.

Eganwp, 2017-01-11:
LOL 43 replies all for nothing...

raamfp, 2017-01-11:
The point here is to help or fix buddy? I dont get it!!! I had a problem. Sorry if u dont like! Thanks everyone.

martinbudden, 2017-01-11:
@raamfp , good to here it's working now. Happy flying.

wind0r, 2017-01-11:
@raamfp No problem :) Nice that we get that stuff sorted out.
2 Bugs fixed and Revolt is now working flawless :)

Maybe it should be written down somewhere that even on a revolt motor 1 is still RR and motor 4 is FL

blckmn, 2017-01-11:
I'll update the wiki so that there's a picture and a bit of a description. Good that it's now working ok.