betaflight/betaflight #1779: BlackBox not working on Airbot/Revo SPI flash.

atomiclama, 2016-12-08:
On this version and some previous versions

# version
# Betaflight/AIRBOTF4 3.1.0 Dec  5 2016 / 13:38:34 (241fb1b)

The BB recording to the onboard SPI flash does not work. 

If I keep hooked up to the configurator when armed I can see the storage used increasing, but after a powercycle the flash indicates empty.

This worked on a previous version, sorry don't know what at the moment, trying to track down the point at which it came in.

atomiclama, 2016-12-08:
Can any git gurus tell me how to set my local repo to a particular commit so that I can trace down where this came in?
Cheers

AndersHoglund, 2016-12-08:
This is only happening when armed? I don't think you can expect data to be retained when cycling power in that state, when recording is going on. 
I can not see any problems with power cycling on my REVO acro clone when dis-armed. BB log data seems to survive.

    # version
    # Betaflight/REVO 3.1.0 Dec  8 2016 / 15:28:19 (5e30866)

atomiclama, 2016-12-08:
Yeah, when armed. I did not know there was another way to get the BB data to record.

In normal flight I get no BB data recorded. Hook up to PC and flash is blank.

On bench and arming I can see status of the flash being used, if I download data it is blank. If I erase flash it goes too quickly.

I'll try the official builds rather than my local build see if that makes a difference.

atomiclama, 2016-12-08:
Ok the problem came in with commit e9083db

It is the change in optimisation to
+OPTIMIZE    = -Ofast

If I switch this back then my BB flash is ok.

A quick way to check this is to do a flash_erase from the command line.

It should take some seconds to complete.

I guess this is a problem due to me using a different version of compilers than the official builds.

I'm using 5.4 2016q3 not sure about the official build. Could someone comment.

AndersHoglund, 2016-12-08:
OK, so your problem is you get no data recorded at all in flight? To me the description sounded as it was data loss when power cycling.... You can start/stop recording on an AUX switch too. Check the modes tab.

Did a quick bench test, power on, arm, wiggle the sticks, wiggle the FC, disarm power off. Connect to BFC and I can see I have data logged. 
Did the same test again, twice. Now with a difference that I powered off when still armed. To my surprice I had data logged in both cases. 

There seems to be a problem with the cli command though. It prints "Era   ", then the board hangs. 

    # flash_erase
    Era
Is this what you see also ??

gcc 5.4 2016q3 should fine, use that too.

AndersHoglund, 2016-12-08:
Ahh, sorry. The CLI command "flash_erase" does not hang the board. It is just slow. Takes about 60s then:

    # flash_erase
    Erasing...
    Done.
But this may still be a UI/cosmetic problem... And the same regardless of optimization settings, tried "-Os" and "-O0" (thru DEBUG=GDB).

AndersHoglund, 2016-12-08:
See PR #1786

atomiclama, 2016-12-09:
Looks like a nice improvement, but don't think it is my problem. I'll give it a try when I can.

Sorry my description was not clear enough. I'll try again and break it down into a few sections.

1. I get no storage of BB data on the flash.
2. flash_info indicates that data has been stored. But after a power cycle flash_info returns no data stored.
3. flash_erase returns immediately.   On a working build it can tale @60sec as you have stated above.

I leaning towards a timing issue, maybe the flash chip on my board is a bit slow. Chip select? As this is driven from SW. Can't gain access to it at the mo to see what it is. Will check later.

If I get a chance I'll hook up a logic and get some traces.

Thanks for confirming that it is not my compiler version. Didn't fancy looking for any asm differences.

AndersHoglund, 2016-12-09:
@atomiclama , could you please flash your board with a recent version that has the problems and post the output from the cli diff command here. You may have things running that I do not have. This might explain why I can not reproduce the problems. But I also only have the striped "Revo Acro" board, so I can not really run everything a fully populated original Revo can. So I might not be of much help here..
The AirbotF4 has at least very small differences. 
Here are my diff:

    # diff
    # version
    # Betaflight/REVO 3.1.0 Dec  8 2016 / 18:58:47 (5e30866)
    # name
    name REVO-EXP
    # resources
    resource BEEPER A00
    resource MOTOR 6 NONE
    # mixer
    # servo
    # servo mix
    # feature
    feature VBAT
    feature LED_STRIP
    feature AIRMODE
    # beeper
    # map
    map TAER1234
    # serial
    serial 2 64 115200 57600 0 115200
    # led
    led 0 4,15:W:AI:0
    led 1 5,15:W:AI:0
    led 2 6,15::AT:0
    led 3 7,15::AT:0
    led 4 8,15::AT:0
    led 5 9,15::AT:0
    led 6 10,15:E:AI:0
    led 7 11,15:E:AI:0
    # color
    # mode_color
    # aux
    aux 0 0 1 1700 2100
    aux 1 1 2 1700 2100
    aux 2 2 2 1300 1700
    aux 3 27 1 1700 2100
    aux 4 12 0 1000 1500
    # adjrange
    adjrange 0 0 3 1900 2000 15 4
    adjrange 1 0 3 1800 1900 18 4
    adjrange 2 0 3 1675 1800 9 4
    adjrange 3 0 3 1550 1675 16 4
    adjrange 4 0 3 1450 1550 19 4
    adjrange 5 0 3 1350 1450 10 4
    adjrange 6 0 3 1225 1350 17 4
    adjrange 7 0 3 1100 1225 20 4
    adjrange 8 0 3 975 1100 11 4
    # rxrange
    # rxfail
    rxfail 4 s 2000
    rxfail 5 s 1500
    rxfail 6 s 1500
    rxfail 7 s 1500
    rxfail 8 s 1500
    rxfail 9 s 1500
    rxfail 10 s 1500
    rxfail 11 s 1500
    # master
    set motor_pwm_protocol = DSHOT150
    set serialrx_provider = SPEK2048
    set blackbox_device = SPIFLASH
    # profile
    profile 0
    rateprofile 0
    # rateprofile
    rateprofile 0
    #

atomiclama, 2016-12-09:
The board I'm using is:-
http://www.banggood.com/Raceflight-Betaflight-CC3D-REVO-F4-STM32F405-Flight-Controller-with-VbatBuzzer-p-1087561.html

All my recent tests have been run with the default config and just issuing a flash_erase from the CLI.
If this takes @60s then the flash is ok and the BB will work, if it returns straight away then fail.

I then config with the following and fly. Yes I know the BB update rate higher than it needs to be ;-)
Not got round to changing that yet. 

feature AIRMODE 

# serial
serial 0 64 115200 57600 0 115200

# aux
aux 0 0 0 1800 2100

set min_check = 1010
set motor_pwm_protocol = DSHOT300
set serialrx_provider = IBUS
set yaw_motor_direction = -1
set acc_hardware = NONE
set pid_process_denom = 1
set blackbox_device = SPIFLASH

set pid_at_min_throttle = OFF

AndersHoglund, 2016-12-09:
Is that the full diff output?  
"... the BB update rate higher...". So, what rate is it? 4K or maybe 2k with denom=1?   Mine is 2K by default with reset config.... First thing I would try is to lower the BB data rate.
There is also the possibility your board is broken or poorly designed.

AndersHoglund, 2016-12-09:
Sorry, can not reproduce. 
Tried your method of resetting to default config and the cli flash_erase cmd still works just fine. Tested with 2k (default) and 4k BB data rate. Also with PR #1786 integrated, no difference.

Some one else that has an AirbotF4 or this BG Revo clone will have to make an attempt of reproducing this.

AndersHoglund, 2016-12-09:
One more thing, have you tried with any Jenkins builds? Just to rule out any possible problems with your compiler installation and build env. My last tests was done with this build:
http://andwho.sytes.net:8080/job/BorisB_BetaFlight/798/

I also did a quick test with both the AIRBOTF4 and REVO builds, both on the same RTFQ Revo Acro board. Same result. all OK with default config and flash_erase cmd.

atomiclama, 2016-12-09:
BB rate does not matter, as the flash_erase is done disarmed from the CLI. FWIW it can run at 8K on a working build.
The board is not broken as it works on builds with no fast optimisation.
Poorly designed maybe :-) But as it is a clone should be reasonable, the manufacture looks good, solder joints etc.

I will be trying a jenkins build when I get some time at home. Unfortunately it's blocked at work.

Can you tell me which flash chip is on your builds? so I can compare with mine at some point.

Oh yeah, thanks for spending the time.

AndersHoglund, 2016-12-09:
Not sure I can see all chip markings clearly. Could be a Micron device:

    25Q128A
    13E40

atomiclama, 2016-12-10:
Jenkins builds fail as well. 
Enjoying flying too much at the mo, will take it apart and have a look next week.

blckmn, 2016-12-11:
Looks like REVOLT BB logging might also be in a non-working state.

atomiclama, 2016-12-12:
Ok managed some time with the logic.

This is the timing from a failed build.
[image]

atomiclama, 2016-12-12:
Thinking about this. It's not the delay that is being added by the NOP but rather it is forcing the compiler to  not optimise as much.

AndersHoglund, 2016-12-12:
As it seems to be the back-to-back CS toggling operations, it would probably be better timing with this order:

    #define DISABLE_M25P16 IOHi(m25p16CsPin); __NOP();
    #define __NOP(); ENABLE_M25P16 IOLo(m25p16CsPin);

Not sure I like controlling timing this way. But as I have no better suggestion, I think this would be good enough. 

One other ting I noticed; Even though the "flash_m2516.c" are NOT part of the "SPEED_OPTIMIZED_SRC" it is still compiled with -Ofast. The Makefile has very strange logic, and does things like this, in some cases:

    OPTIMISE_SIZE         = -Ofast

The Makefile really needs some cleanup here, not really logical nor maintainable. But that is another issue.. But fixing that would probably also hide/fix flash timing issues.. WAM (What A Mess).

atomiclama, 2016-12-13:
Ok created a PR #1819 with the order suggested above

Tested on bench with flash_erase command. Get this waveform

[image]

atomiclama, 2016-12-19:
#1819 merged.

closed 2016-12-19 by atomiclama