betaflight/betaflight #1781: FC sends MSP_DISPLAYPORT message when rebooting.

mikeller, 2016-12-08:
@martinbudden: This occurs when connected to the Betaflight configurator, when rebooting for 'Save & Reboot' or when leaving the CLI.

jflyper, 2016-12-08:
@mikeller Just once? Did you capture the parameters for the MSP_DISPLAYPORT message?

mikeller, 2016-12-08:
No I didn't. And it does not reproduce with the BLUEJAYF4 that I have here. Will repro tonight at home.

jflyper, 2016-12-09:
Some thoughts.
- MSP_DISPLAYPORT messages are suppressed for VCP connections:
msp/msp_serial.c:L233-236 (mspSerialPush)
```
233         // XXX Kludge!!! Avoid zombie VCP port (avoid VCP entirely for now)
234         if (mspPort->port->identifier == SERIAL_PORT_USB_VCP) {
235             continue;
236         }
```
So the bug should be differently between VCP and non-VCP MSP connections.

- If displayPortRelease is called during shutdown, it may emit close sub-command.

mikeller, 2016-12-09:
Some answers:
- the unsolicited messages only appeared on non-VCP boards. BLUEJAYF4 (VCP) doesn't show them;
- sounds plausible that it happens on shutdown.

I am also not insinuating that the messages are a bad thing per se - i only discovered them because I was tracking down some other messages that weren't properly recognised due to an incomplete rename in the configurator. Simplest case would probably be to just add them to the list of recognised messages in the configurator, and handle them by ignoring them. I just wanted to make sure that this is not the symptom of some condition in the firmware that needs fixing.

mikeller, 2016-12-10:
@jflyper: I finally got around to capturing the `MSP_DISPLAYPORT` message that was sent to the configurator, and the data is `[2]`.

jflyper, 2016-12-10:
That's screen clear. Narrowed down a little bit.

martinbudden, 2016-12-10:
OK, I think the guilty code is here: https://github.com/betaflight/betaflight/blob/master/src/main/drivers/display.c#L99-L106

The `displayInit` function rather over enthusiastically calls `clearScreen`. My fault - I put that in. The lines:
```
 instance->vTable->clearScreen(instance);
 instance->cleared = true;
```
should simply be removed from that function.

jflyper, 2016-12-10:
Ah, that's it.

But I think the configurator should gracefully ignore the MSP_DISPLAYPORT messages; all MSP_DISPLAYPORT messages are delivered to all MSP enabled ports. (We need a new switch to suppress injected messages.)

mikeller, 2016-12-12:
https://github.com/betaflight/betaflight-configurator/pull/362