betaflight/betaflight #1782: Add vtx power support

Linjieqiang, 2016-12-08:
The RTC6705 chip datasheet says it can change rf power by writing different register value.Currently the befaflight code can change vtx power between 25mw and 600mw.But I think it can provide various of choise,such as 25mw,200mw,400mw,600mw even 800mw by the code.I will make a pr in a few days to change vtx power.Does anybody has any idea?
[image]

jflyper, 2016-12-08:
@Linjieqiang Make sure that final output power agree with the intended power.

HaseebNain, 2016-12-08:
We've been looking into this as well, the issue is that the datasheet we have is incredibly difficult to read. The registers have names but those names and the impact of the control bits don't have documentation showing their effect.

Makes it trial and error trying to guess the effect of changing values of the registers.

Is there a better documented source than this?
[RTC6705-DST-001 Datasheet.pdf](https://github.com/betaflight/betaflight/files/640484/RTC6705-DST-001.Datasheet.pdf)

Linjieqiang, 2016-12-09:
I checked it with spectrum analyzer but had a little wrong and different vtx had different parameters with 25mw,200mw,400mw,600mw.I didn't find out what's right way to change it.

jflyper, 2016-12-09:
@Linjieqiang It depends on the supply voltage of the RF module at least. Take FX758-2 for example. Even with default register values, it would transmit at different powers with different supply voltages.

Linjieqiang, 2016-12-09:
Yeah.@jflyper But I want to change it by writing reigster rather than change different supply voltages.The supply voltage with my vtx on my fc was 5.0V and it showed 28dBm which means the 600MW but the  temperature was too high.So I hope it can be changed only with writing reigster value.

jflyper, 2016-12-09:
@Linjieqiang You can do whatever to lower your RF module temperature; tweaking PA gain, changing supply voltage to 3V3.

But, you are proposing a consolidated and consistent way of switching output powers among different bleed of vtx, aren't you?

Linjieqiang, 2016-12-09:
Indeed,there is a little error between the final output power and the intended power.So different vtx has different values.

phobos-, 2016-12-09:
It all depends on the vtx you're using. RTC6705 is able to provide max 13dBm (20mW) with it's built in internal PA according to the datasheet. Most VTX modules with power greater than 25mW have external PA chip (most of the time it's RTC6671, but not always).

There is no way to set power output to more than 25mW on a 25mW VTX module as it has no external PA.

Having said that, I think the power setting shouldn't have hardcoded power values , instead maybe percentage values, like 10%, 20%... 100%.

P.S. I'm trying to connect my VTX to an F3 MCU using SINGULARITY target. How to connect RTC6705 to a hardware SPI? Is that mapping correct?
SPIDATA = MOSI PIN
SPICLK = CLK PIN
SPILE = SS PIN

Linjieqiang, 2016-12-10:
@phobos- Forgive me for coming so late. The RTC6705 which connected to hardware SPI is right!Your idea which uses percentage values sounds great.I will have a try.

DanNixon, 2016-12-10:
Not sure if I really like the idea of percentage power values, when the rest of the hobby measures power in mW measuring in % is only going to cause confusion.
I'm not entirely sure what the advantage is either. In that case why not just have an integer power in mW and the device uses the nearest possible power that is not above the set value?

jflyper, 2016-12-10:
I don't like the percentage either.

@phobos Few comments.
- RTC6705's SPI is not SPI, but SPI-ish. The worst behavior is that it doesn't honor SPILE as SS (It always listens to CLK/DATA, so it can't share a SPI with other SPI device).
- It may not be a good idea to dedicate a scarce resource like SPI for devices that is accessed few times per power cycle; bit-banging with less useful gpio pin is more appropriate.
- Refer to SIRINFPV target.

phobos-, 2016-12-10:
@DanNixon @jflyper In that case it would require a hardcoded or user configurable maximum power value in mW for each target with built in VTX. Calculating the mW value based on that constant and a register value would be a piece of cake. You could input an integer value of mW between 0 and max, i.e 457mW for a 600mW VTX module, and it would set apropriate values in the register.

But still it wouldn't reflect the truth because of voltage differences, non-linearities of the output at different register values, etc, etc. It would be a good enough approximation in my opinion though.

@jflyper agreed, thanks. The worst thing is that hardware SPI doesn't allow for reading register values natively from this particular chip, so I would have to code bit banging just for that anyway.

jflyper, 2016-12-12:
For the power indices, I would go for symbolic, such as OFF, LOW, MID, HIGH, EXTRA.

mikeller, 2016-12-12:
I'd go for MIN and MAX for minimum / maximum power.

jflyper, 2016-12-12:
Ah, good word choice. The EXTRA was out of desperation.

OFF MIN LOW HIGH MAX

Linjieqiang, 2016-12-12:
Great."OFF MIN LOW HIGH MAX" sounds great.

DanNixon, 2016-12-12:
So if I have a 200mW VTX and set it to LOW, what is my output power?
Can you see the issue I have with not measuring power in mW?

jflyper, 2016-12-12:
@DanNixon 
Since it is impossible to detect how final stage PA is configured, something like OFF, MIN, LOW, HIGH, MAX is the best compromise. We can, of course, have a "vtx_device_type" cli variable which select a table of actual output power levels (or even have a vtx_power_level_0, vtx_power_level_1, ... array), provided someone take responsibility on the characteristics of the particular vtx device. For FC integrated VTX-RF, board maintainer can do it. For random externally controllable vtx, I don't know.

mikeller, 2016-12-12:
Agreed. But in the end the (supposed) output power of the RFM is not really what's interesting, it's the EIRP, which depends on a lot more than the RFM. At least with relative description you get a feel of 'HIGH is more than LOW'.

DanNixon, 2016-12-12:
@jflyper Right, I though the RTC6705 was only really used when integrated onto the FC board, in which case knowing the PA configuration is not an issue.
That being the case then yes, the generic names are probably the easiest option.

jflyper, 2016-12-12:
@Linjieqiang So, are constants in the first comment derives something closer to what you want?
I noticed values for 400mW and 600mW are both 0x4FBD.

Linjieqiang, 2016-12-13:
I will base the  spectrum analyzer to adjust this value later.@jflyper

Linjieqiang, 2016-12-16:
Three points:
- Using "OFF, MIN, LOW, HIGH, MAX " to represent the rf power for users.
- Add cli item to adjust it truely.
- Add CMS item to adjust it in OSD for users.
- What else?

But I want to know the meaning of the PA control register7 in RTC6705 data sheet.Anyone knows?@savaga Can you give me some help about how to adjust the rf power well?

Linjieqiang, 2016-12-16:
OK.I tried to commit firstly on this: https://github.com/betaflight/betaflight/pull/1839

hydra, 2016-12-28:
See Cleanflight codebase, it already has OFF (soft power)/LOW/HIGH support.

See also #1912 - in particular `rtc6705SetRFPower`

Linjieqiang, 2016-12-29:
@hydra OK,I will pay attention it and change my code to adjust it.Thanks.

tigert, 2017-02-02:
A case: a bunch of folks flying FPV together . some fly a racetrack close by requiring low mW to avoid interference, but someone else wants to go farther away with a fixed wing, for example. 

I wonder what happens during a power switch, does the video go blank, and for how long? Would it be feasible to try to link the VTX transmit power to, say, RSSI strength, so when you fly close to yourself, it would lower the mW (and be nicer to your flying buddies) but when you go far out, it would crank it up accordingly?

Same could happen with arming. I don't know if it does this already, but optionally it would be nice to have "pit mode" minimal mW when unarmed, and crank it up when you actually arm and want to go flying. 

This could perhaps be a mode you can activate, because I guess there are cases when you want something else.

jflyper, 2017-02-02:
@tigert There are lots of interesting enhancements that we can do with FC controlling VTX, and yes, RSSI based output adjustment and arming triggered pit mode release has been talked about. And some of the most wanted features will be implemented in the future. However, for the moment, we are focusing on consolidating multiple VTX mechanisms so that we can use the consolidated API with more advanced features such as the ones you mentioned.

In the mean while, why don't you submit a separate feature request, so people can discuss on it.

digaus, 2017-03-15:
Hey guys, just want to know if it is already possible to change the power and frequency of a VTX which is connected via SPI with the betaflight lua script? I am asking because I plan to buy the Fly FishTower f4 (Fishdrone F4) which has the VTX connected via SPI. It mentions that it supports VTX configuration via OSD but does not say anything about lua script.

jflyper, 2017-03-15:
@digaus See answer to your another question.

digaus, 2017-03-15:
Yeah my bad then ... thought one could just use parts from Tramp/Unify and combine them with this :/
Will look into this aswell if I find the time.

jflyper, 2017-03-15:
@digaus The code is almost done. Waiting for power control values from fishdrone. If they are not available, I will just make it low and high (plus off for NEO).

digaus, 2017-03-15:
That is really nice to hear! Yeah some more powerlevels would be nice, at least pidmode, low (25mw) and max

Linjieqiang, 2017-03-16:
I will take a look with your code.@jflyper But where's the code?On which branch?

jflyper, 2017-03-17:
@Linjieqiang I'm referring to your PR #1839

Linjieqiang, 2017-03-17:
OK.Thanks. @jflyper

fncisco, 2017-03-17:
In 3.1.6/vtx_soft_spi_rtc6705.h
#define DP_5G_MASK                0x4000 //7000
#define PA5G_BS_MASK            0x0E00
#define PD_Q5G_MASK             0x0040
#define PD_Q5G_MASK_OFF     0x0000
#define QI_5G_MASK                0x0038 
#define PA_BS_MASK                0x0005 //0007
#define POWER_0_MASK          0x0000  //25mW
#define POWER_1_MASK          0x0080  //200mW
#define POWER_2_MASK          0x0100  //400mW
#define POWER_3_MASK          0x0180  //600mW?800mW?

fncisco, 2017-03-17:
In 3.1.6/vtx_soft_spi_rtc6705.c
void rtc6705_soft_spi_set_rf_power(uint8_t reduce_power)
{
    //rtc6705_write_register(7, (reduce_power ? (PA_CONTROL_DEFAULT | PD_Q5G_MASK) & (~(PA5G_PW_MASK | PA5G_BS_MASK)) : PA_CONTROL_DEFAULT));
    switch(reduce_power){
        case 0:
            rtc6705_write_register(7, ((POWER_0_MASK | PA_CONTROL_DEFAULT) & ~PD_Q5G_MASK));
        break;
        case 1:
            rtc6705_write_register(7, (POWER_0_MASK | PA_CONTROL_DEFAULT));
        break;
        case 2:
            rtc6705_write_register(7, (POWER_1_MASK | PA_CONTROL_DEFAULT));
        break;
        case 3:
            rtc6705_write_register(7, (POWER_2_MASK | PA_CONTROL_DEFAULT));
        break;
        case 4:
            rtc6705_write_register(7, (POWER_3_MASK | PA_CONTROL_DEFAULT));
        break;
    }
}

fncisco, 2017-03-17:
I have added other code to cms, osd, cli and fcinit, to get this 5 options active, just rest test each power command to verify the real change on output power.
if some one want test it, I'll put a clone on fncisco/betaflight

Linjieqiang, 2017-03-18:
@fncisco Looks nice.But it needs to test.

jflyper, 2017-03-18:
@fncisco What the theory behind POWERx_MASK values?

It looks as `PA5G_PW` decides output power when `PD_Q5G` is off, and sounds sensible...

@Linjieqiang Can you test the values?

fncisco, 2017-03-18:
@jflyper I'm already test with help os Guo Wei (Exuav) and this flags are based on #fail rtc6075 datasheet, they don't work, but we tested others Values and get some rasonable new ones. I'm already integrate on 3.16 here is code: https://www.dropbox.com/s/qwigqybxa9embve/betaflight-3.1.6.7z?dl=0

jflyper, 2017-03-19:
@here 

I'm trying to get a theory behind the experimentally obtained PA control register values, so that they can be safe to different kind of RF modules based on RTC6705. It is not just 'It works for this hardware'.

While 4 different values to PA5G_PW controlling final output power sounds reasonable, but I'm still skeptical about the output power values, e.g., 25, 200, 400 and 600/800mW.

I think I'll just give MIN, LOW, HIGH, MAX as discussed above.

digaus, 2017-03-19:
@jflyper

What about pidmode (OFF)?^^

jflyper, 2017-03-24:
@digaus I think you are mixing several things up.
- PIT mode
My understanding is that PIT mode is different from RF power selection, and should be controlled differently. For example, some want auto-PIT-out upon arming. If PIT was one of the power selections, this feature can not be implemented.

- Power OFF
Power off is different from PIT mode, because it may turn off the VTX completely and make you blinded. A stick command may be created to turn it on again, but that's going to be a future feature.

woutput, 2018-01-11:
Please note that at least in Europe (I could only quickly find this low quality reference: http://forum.freeflysystems.com/index.php?threads/video-downlink-tx-power-what-is-legal.2364/), maximum legal transmission power is limited to 25 mW. Please consider this; MIN/MAX is insufficient to cover this.

mikeller, 2018-06-03:
@Linjieqiang: I think this has been addressed. Can this be closed?

Linjieqiang, 2018-06-04:
Closed done!Thanks @mikeller

closed 2018-06-04 by Linjieqiang