betaflight/betaflight #1784: BF 3.0.1 board align issue on target motolab, seems broken

moriwaky, 2016-12-08:
Hi, I have a MotoF3 Flight Control Board / PDB and I have to use align board on yaw axis 180 degrees. Everything seem OK on configurator when you move the quad it moves the same on the screen. But when you try to take off the quad crashes very violenty like when one motor is plugged on the wrong conector. The quad flies fine when I have not select "align the board" but I have to inverter the elevon and aliron on the remoted control to fly fine. Can anyone take a look?

mikeller, 2016-12-08:
I think you are confusing board alignment with RC channel mapping. These two control different things. Use  'Channel Map' in the Betaflight configurator to set up your RC channel mapping:

[image]

moriwaky, 2016-12-08:
Maybe because my english  is not good you has missunderstand me,  I am not confusing channel mapping  with board aligment I just said what I have to do  with a motolab fc mounted backguards (180 degrees yaw axis) on my frame to get it fly with out crash it because the como andas "board aling"  seems broken on firmware

mikeller, 2016-12-08:
Ah now I see. Have you tried setting -180 degrees instead of 180 degrees?

moriwaky, 2016-12-09:
Nope,  I did try only 180. Tomorrow I will try minus 180 and I will post results. Thanks for the idea

moriwaky, 2016-12-09:
Hi Mikeller, I have tried the minus 180 degrees rotation setting and same result crash very hard just after arming, so from my point of view sominthing is broken here in firmware.

AndersHoglund, 2016-12-09:
You have at least three separate things mixed up.

1. Board orientation. Make sure the  image in BFC move in the same way as the physical quad when moved. Adjust "Board and Sensor Alignment settings" if needed. 

2. Rx Channel mapping. Make sure the correct throttle/yaw/pitch/roll bar move when moving the corresponding Tx sticks. Adjust channel mapping if needed.

3. Channel direction. Make sure channel bars in BFC move to the right when sticks are moved to the right and/or moved up. Adjust Tx channel reversal if needed.

All three are completely independent and has nothing with each other to do, check and adjust them separately.

Then also make sure Tx sticks center at 1500 and endpoints are 1000 and 2000.

moriwaky, 2016-12-09:
hi @AndersHoglund  thanks for your help but seriusly there is nothing wrong with my setup this isn´t my first quad with betaflight and I have use the alignment board command many time in the past with other quads.

As I said on my first post if a choose to align the board 180 or -180 degrees the quad and the movements of the quad model in the GUI match perfect, but when you try to take off it crashes like when you plug the wrong motor on the wrong sport on the board.

My channel mapping is correct I only metion that because the solution I found to fly my quad without using the alignment command is to inverting elevon and aliron just because my flight controller is mounted  looking back

My stick are perfect center at 1500 and my midcommand is 1500 I have check it on the GUI

Can anyone test what I am saying? you just have to use a motolab flight controller and use  board alignenmet command. Be aware it will crash inmediatly after arming

mikeller, 2016-12-10:
@moriwaky What target are you using? Your description makes it sound like the orientation of the gyro on the board is not what the firmware thinks it should be. Potentially there's another target that has the gyro alignment defined the way your board needs it to be.

moriwaky, 2016-12-10:
Hi @mikeller  it makes sense what are you saying. My FC is motof3 flight controller PDB.

this one: http://impulserc.com/motof3-flight-controller-pdb

can I solve the problem by my self or is it need to change firmware?

thanks!

mikeller, 2016-12-10:
From what you've said above, you managed to find a workaround that works for the moment.

If this board has indeed the gyro mounted differently to other boards that use the same target, then we should see if we can detect this board, in the firmware, and configure it accordingly. This is best left to somebody who actually has this board, so they can test.

moriwaky, 2016-12-10:
Ok, perfect I normally use a little bit of roll_yaw_cam_mix_degrees, but with my solution it works on the other way because the mix of the controls with the bad board orientation. So I will wait until somenone can find a solution for this board. But I would like to warnign everybody of this posible bug with the gyro of this board because the first time I crash was very very scary and totally unexpected since everything was fine in the GUI, and in the setup of the quad.

thanks @mikeller

mikeller, 2016-12-10:
In the mean time you should be able to build your own version of the firmware that fixes this for your board.

Also, if you had bench tested your quad with props off, you should have noticed that the motors spinning up when lifting the quad by hand and moving it around were not the low but the high ones, and realised that something was amiss...

moriwaky, 2016-12-10:
My coding skills are far from good to make me my self a special firmware for this target, I am thinking maybe I can solve with some "magic" using a custom motor mixer setup.

moriwaky, 2016-12-10:
UPDATE: I have no luck with mixer custom I have tried every single posiblity and nothing works as expected

If this is the normal mix for a quadx:

mmix 0 1.0, 1.0, -1.0, -1.0
mmix 1 1.0, -1.0, -1.0, 1.0
mmix 2 1.0, -1.0, 1.0, -1.0
mmix 3 1.0, 1.0, 1.0, 1.0

I have tried every sigle rotation on yaw axis for the motors:

90 degrees to the rigth

mmix 3 1.0, 1.0, -1.0, -1.0
mmix 0 1.0, -1.0, -1.0, 1.0
mmix 1 1.0, -1.0, 1.0, -1.0
mmix 2 1.0, 1.0, 1.0, 1.0

180 degrees
mmix 2 1.0, 1.0, -1.0, -1.0
mmix 3 1.0, -1.0, -1.0, 1.0
mmix 0 1.0, -1.0, 1.0, -1.0
mmix 1 1.0, 1.0, 1.0, 1.0

270 degrees to the right

mmix 1 1.0, 1.0, -1.0, -1.0
mmix 2 1.0, -1.0, -1.0, 1.0
mmix 3 1.0, -1.0, 1.0, -1.0
mmix 0 1.0, 1.0, 1.0, 1.0

inverted 180

mmix 3 1.0, 1.0, -1.0, -1.0
mmix 2 1.0, -1.0, -1.0, 1.0
mmix 1 1.0, -1.0, 1.0, -1.0
mmix 0 1.0, 1.0, 1.0, 1.0

inverted 90 degrees

mmix 1 1.0, 1.0, -1.0, -1.0
mmix 0 1.0, -1.0, -1.0, 1.0
mmix 3 1.0, -1.0, 1.0, -1.0
mmix 2 1.0, 1.0, 1.0, 1.0

cross invertion

mmix 2 1.0, 1.0, -1.0, -1.0
mmix 3 1.0, -1.0, -1.0, 1.0
mmix 0 1.0, -1.0, 1.0, -1.0
mmix 1 1.0, 1.0, 1.0, 1.0

nonen worked so I think I will fly with my solution and wait if can be fixed in the future. The main 
disadvantage of my solution is I can´t use  roll_yaw_cam_mix_degrees feature anymore because is behavior is inverted as you will expect because the wrong orientation of the board. Everything works fine by the way. 
So many thanks guys for the help.

AndersHoglund, 2016-12-10:
I think you are on the right track now. Motor placement, not board orientation. If not already on BF3.1, update, and use the resource cmd to relocate motor outputs. Motor order is defined in src/main/target/MOTOLAB/target.c , if you want to play with it there.

But first, as I tried to say before, make sure board orientation, channel mapping and channel direction are correct. You do not want to complicate everything with multiple layers of compensations.

AndersHoglund, 2016-12-10:
Strange, I can not find anyone else complaining on board alignment, orientation or motor order here: https://www.rcgroups.com/forums/showthread.php?2537379-MotoLab-Board-Setup-and-Troubleshooting
Only one guy struggling with custom mixer for a board mounted up side down, And aother guy who has done the same mounting, but only needed to adjust board alignment. Start reading here: https://www.rcgroups.com/forums/showthread.php?2537379-MotoLab-Board-Setup-and-Troubleshooting/page82#post34599540
How is your board mounted?

One more check Q: Was this board working OK with other/older CF/BF versions, and the problem started showing up with BF 3.0.1 ? If so, what versions worked?

moriwaky, 2016-12-10:
Hi @AndersHoglund I was wrong when I made my first testing with custom mixer. At the end I get it working with this custom mixer and using board align on yaw axis 180 degrees:

mmix 3 1.000 -1.000 1.000 -1.000
mmix 2 1.000 -1.000 -1.000 1.000
mmix 1 1.000 1.000 1.000 1.000
mmix 0 1.000 1.000 -1.000 -1.000

Maybe you can´t find any  complain in RCgroups because the board I am using is this:
http://impulserc.com/motof3-flight-controller-pdb is not the normal motolab
this board is for one type of frame and maybe nobody else had to mounted like me looking backguards because I am using a custom frame and I need it in this position. 

For me the conclusion is as @mikeller pointed there is a problem with the orientation of the gyro chip on the board and the firmware is not working rigth when using "align board" on cli.

Thanksssssss guys!

jflyper, 2016-12-11:
@moriwaky Wow. the MOTO F3 PDB has no indication of motor positions what so ever. No wonder you struggle through this.

I think you should
(1) Decide which way is the front you want to be (probably based on the lipo input cable and usb connector positions).
(2) Adjust the yaw rotation of the board orientation so that the 3D model in the configurator moves as you move the craft.
I think you have done these, and
(3) Use resource command in cli to move motor positions.

Using mmix to adjust the motor positions would complicate if you ever want to use black box to tune your racer.

MotoLab, 2016-12-13:
The MotoF3 uses the same gyro orientation as all other MotoLab targets. The lipo connector is at the rear side of the board. Motor outputs are on the corners for normal orientation, and will require cross connection or remapping for a reversed installation.

mikeller, 2016-12-13:
@moriwaky: I totally concur with @jflyper's comment. If you are mounting your FC in a non-default orientation (which is something you probably should have mentioned to start with), then adjusting the board orientation, and then using resource remapping to get the motor outputs into convenient places is the way to go. This way, your board will act 'natural' in all respects.

moriwaky, 2016-12-13:
Hey @mikeller  I was trying to say that had to use aling yaw rotation to 180 because my board was not mounted in the default position why I should use the aling board if not?  and I said this in my first line of my first post here so...

@MotoLab thanks for the info, I have other quads with other flight controller as spracingf3 and I mounted on not default(🙃@mikeller)  position and never had to remap motor positions,  so because of that i though some thing was wrong on firmware 

Anyway thanks to everyone who help me solving my problem

borisbstyle, 2016-12-19:
Closing issue as totally not firmware related it seems...,.

closed 2016-12-19 by borisbstyle