betaflight/betaflight #1787: IO remapping capability breaks flying wing configuration

rafl, 2016-12-09:
Trying revision 5f47fc31 on a flying wing, I've noticed that the two servo outputs PWM 3 and 4 no longer provide pulses and instead are constantly at ground level. This problem is easy to reproduce by flashing betaflight onto a flight controller doing a full erase, switching to the flying wing mixer, and sampling the outputs.

This is a change in behaviour since betaflight 3.0.1, where the same procedure would result in PWM pulses on two of the outputs. Bisecting the problem leads to commit 46a6e560.

This seems to be an issue across different flight controllers. I've been able to observe it on the following targets: SPRACINGF3, SPRACINGF3EVO, OMNIBUS, and OMNIBUSF4.

At first I assumed I'd be able to work around this by using the new resource configuration to have the appropriate output pins mapped to a servo rather than a motor using commands along the lines of `resource motor 3 none` and `resource servo 3 a02`, but that doesn't seem to change the behaviour at all and makes this appear to be a real regression as opposed to a change in the configuration required for flying wing mixing.

mikeller, 2016-12-09:
1. Have you rebooted the FC after changing the resources with the `resource` command?
2. After rebooting, did `resource list` show the servo on A02? If not, did you try freeing whatever other resource was shown on A02?

For debugging' sake: What mixer did you select? (Mixer selection determines what resources are allocated.)

rafl, 2016-12-09:
I've used the `save` command after changing resources, which I believe will reboot the device. I'm using the FLYING_WING mixer.

Here's what things look like for SPRACINGF3EVO:

```
# resources after flashing + erase
# resource
resource BEEPER C15
resource MOTOR 1 A00
resource MOTOR 2 A01
resource MOTOR 3 A02
resource MOTOR 4 A03
resource MOTOR 5 A06
resource MOTOR 6 A07
resource MOTOR 7 B00
resource MOTOR 8 B01
resource MOTOR 9 B10
resource MOTOR 10 B11
resource PPM A15
resource LED_STRIP A08
# resource list
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 1
A01: MOTOR 2
A02: MOTOR 3
A03: MOTOR 4
A04: ADC_BATT 
A05: ADC_CURR 
A06: FREE 
A07: FREE 
A08: TRANSPONDER 
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: FREE 
A12: FREE 
A13: FREE 
A14: FREE 
A15: PPM 
B00: FREE 
B01: FREE 
B02: ADC_RSSI 
B03: SPI_SCK 1
B04: SPI_MISO 1
B05: SPI_MOSI 1
B06: I2C_SDA 1
B07: I2C_SCL 1
B08: LED 1
B09: MPU_CS 
B10: FREE 
B11: FREE 
B12: SDCARD_CS 
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: MPU_EXTI 
C14: SDCARD_DETECT 
C15: BEEPER 
F00: FREE 
F01: FREE 
F04: FREE 

Currently active DMA:
--------------------
DMA1 Channel 1: FREE
DMA1 Channel 2: TRANSPONDER
DMA1 Channel 3: FREE
DMA1 Channel 4: FREE
DMA1 Channel 5: FREE
DMA1 Channel 6: FREE
DMA1 Channel 7: FREE
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE

# changing to flying wing mixer
# mixer flying_wing
Mixer: FLYING_WING

# resource motor 3 none
Resource is freed!

# resource motor 4 none
Resource is freed!

# resource servo 1 a02
Resource is set to A02!

# resource servo 2 a03
Resource is set to A03!

# save

# resource
resource BEEPER C15
resource MOTOR 1 A00
resource MOTOR 2 A01
resource MOTOR 5 A06
resource MOTOR 6 A07
resource MOTOR 7 B00
resource MOTOR 8 B01
resource MOTOR 9 B10
resource MOTOR 10 B11
resource SERVO 1 A02
resource SERVO 2 A03
resource PPM A15
resource LED_STRIP A08

# resource list
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 1
A01: FREE 
A02: SERVO 1
A03: SERVO 2
A04: ADC_BATT 
A05: ADC_CURR 
A06: FREE 
A07: FREE 
A08: TRANSPONDER 
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: FREE 
A12: FREE 
A13: FREE 
A14: FREE 
A15: PPM 
B00: FREE 
B01: FREE 
B02: ADC_RSSI 
B03: SPI_SCK 1
B04: SPI_MISO 1
B05: SPI_MOSI 1
B06: I2C_SDA 1
B07: I2C_SCL 1
B08: LED 1
B09: MPU_CS 
B10: FREE 
B11: FREE 
B12: SDCARD_CS 
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: MPU_EXTI 
C14: SDCARD_DETECT 
C15: BEEPER 
F00: FREE 
F01: FREE 
F04: FREE 

Currently active DMA:
--------------------
DMA1 Channel 1: FREE
DMA1 Channel 2: TRANSPONDER
DMA1 Channel 3: FREE
DMA1 Channel 4: FREE
DMA1 Channel 5: FREE
DMA1 Channel 6: FREE
DMA1 Channel 7: FREE
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE
```

mikeller, 2016-12-09:
One more thing that comes to my mind is DMA conflict. What ESC protocol have you configured for the motors? Can you try switching it to PWM and see if the servos work then?

rafl, 2016-12-09:
I think your suspicion might've been right. I've been using OneShot 125 as the ESC protocol when I was having issues. The servo outputs started behaving the way I expected them to after switching to PWM.

jflyper, 2016-12-09:
Can you paste the result of resource and resource list commands for the working version of configuration?

mikeller, 2016-12-09:
Ok. Looks like it's a resource conflict then. Sharing timers between Oneshot / Multishot protocols and servo PWM doesn't work (and didn't work before 3.1). Without knowing what board you're using it's impossible to tell you what will work, but in general the same pin assignments that worked for 3.0.1 should work for 3.1, since the hardware constraints are the same. Also, if you look at `src/main/{target}/target.c` you should be able to figure out how to set up your motors and servos to use different timers.

jflyper, 2016-12-09:
@mikeller I'm a bit more clever now :)

mikeller, 2016-12-09:
@jflyper You wanna share? :-P

rafl, 2016-12-09:
Here's what things look like in a working state on SPRACINGF3EVO, with the motor protocol set to PWM.

```
# resource
resource BEEPER C15
resource MOTOR 1 A00
resource MOTOR 2 A01
resource MOTOR 5 A06
resource MOTOR 6 A07
resource MOTOR 7 B00
resource MOTOR 8 B01
resource MOTOR 9 B10
resource MOTOR 10 B11
resource SERVO 1 A02
resource SERVO 2 A03
resource PPM A15
resource LED_STRIP A08

# resource list
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 1
A01: FREE 
A02: SERVO 1
A03: SERVO 2
A04: ADC_BATT 
A05: ADC_CURR 
A06: FREE 
A07: FREE 
A08: TRANSPONDER 
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: FREE 
A12: FREE 
A13: FREE 
A14: FREE 
A15: PPM 
B00: FREE 
B01: FREE 
B02: ADC_RSSI 
B03: SPI_SCK 1
B04: SPI_MISO 1
B05: SPI_MOSI 1
B06: I2C_SDA 1
B07: I2C_SCL 1
B08: LED 1
B09: MPU_CS 
B10: FREE 
B11: FREE 
B12: SDCARD_CS 
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: MPU_EXTI 
C14: SDCARD_DETECT 
C15: BEEPER 
F00: FREE 
F01: FREE 
F04: FREE 

Currently active DMA:
--------------------
DMA1 Channel 1: FREE
DMA1 Channel 2: TRANSPONDER
DMA1 Channel 3: FREE
DMA1 Channel 4: FREE
DMA1 Channel 5: FREE
DMA1 Channel 6: FREE
DMA1 Channel 7: FREE
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE
```

I'm not sure how to tell which resources are conflicting, or how to prevent them from conflicting.

I hope betaflight won't require its users to have this knowledge in upcoming versions. Having a simple `mixer flying_wing` do the right thing again regardless of the ESC protocol used would be fantastic.

mikeller, 2016-12-09:
Looking at `src/main/target/SPRACINGF3EVO/target.c`, putting your motor onto A03 (TIM15), and your servos onto A00 and A01 (TIM2) should work. Alternatively, assuming that ultrafast motor response is not crucial for a flying wing, you could just enable PWM on your ESC and use it the way you have confirmed it working.

We are still in the process of working out how to make resource assignment easy for users, and finding a way to represent / indicate / help solve resource conflicts will be part of that. Unfortunately, before 3.1, resource conflicts were managed by having a hardcoded list of 'if mixer X and board Y, use resource allocation Z', which was hard to maintain, and not sustainable considering the large number of different boards we support.

blckmn, 2016-12-09:
yes @mikeller is right. Having resource defaults for every board and mixer combination is an exponentially difficult proposition, especially when only a few users might use some of the edge combos. At the moment Quad is the default - with the option for users to configure as they wish.

In the future we will be able to assign the which of the valid timers to be used for each pin also which will mean even more possibilities for users (and the associated support quagmire). This will mean we will need a simplified GUI way of assigning resources - BUT - will make it so much more powerful for users to setup their boards how they want - AND - likely will reduce the number of targets required.

borisbstyle, 2016-12-20:
Going to close this issue, but better code guarding against unsupported configurations would be nice to have.

closed 2016-12-20 by borisbstyle