betaflight/betaflight #1789: gyro update frequency broken

TheAngularity, 2016-12-10:
@martinbudden 
i think your latest #1757 brakes the change of gyro update frequency ... No matter what I set in configurator it always runs with 1khz :(

cli outputs:
```
Entering CLI Mode, type 'exit' to return, or 'help'

# version
# Betaflight/SPRACINGF3EVO 3.1.0 Dec 10 2016 / 08:42:13 (6a363c8)

# status
System Uptime: 15 seconds, Voltage: 8 * 0.1V (0S battery - NOT PRESENT), CPU:2%
CPU Clock=72MHz, GYRO=MPU6500
Stack size: 2048, Stack address: 0x10002000
Cycle Time: 1010, I2C Errors: 1, config size: 1560
SD card: None inserted

# tasks
Task list          rate/hz  max/us  avg/us maxload avgload     total/ms
00 - (       SYSTEM)      9       6       1    0.5%    0.5%         0
01 - (     GYRO/PID)   1000     556      83   56.1%    8.8%      1369
04 - (           RX)    101      54      47    1.0%    0.9%        73
05 - (       SERIAL)     98   14042       3  138.1%    0.5%       212
06 - (      BATTERY)     49      11       3    0.5%    0.5%         2
07 - (       BEEPER)     99       9       1    0.5%    0.5%         2
10 - (     LEDSTRIP)     99     284     167    3.3%    2.1%       252
13 - (          CMS)     59       7       2    0.5%    0.5%         1
RX Check Function                10       5                         7
Total (excluding SERIAL)                      62.4%   13.8%

# 
```

here my diff:
```
Entering CLI Mode, type 'exit' to return, or 'help'

# diff

# version
# Betaflight/SPRACINGF3EVO 3.1.0 Dec 10 2016 / 08:42:13 (6a363c8)

# name
name HX200

# resources

# mixer

# servo

# servo mix

# feature
feature -RX_PPM
feature -TELEMETRY
feature -CURRENT_METER
feature -RSSI_ADC
feature -TRANSPONDER
feature RX_SERIAL
feature LED_STRIP
feature AIRMODE

# beeper

# map
map TAER1234

# serial
serial 0 0 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# led
led 0 11,10::AW:0
led 1 10,10::AW:0
led 2 9,10::AW:0
led 3 8,10::AW:0
led 4 7,10::AW:0
led 5 6,10::AW:0
led 6 5,10::AW:0
led 7 4,10::AW:0
led 8 9,5::CW:6
led 9 8,5::CW:6
led 10 7,5::CW:6
led 11 6,5::CW:6
led 12 6,2::CW:6
led 13 7,2::CW:6
led 14 8,2::CW:6
led 15 9,2::CW:6

# color

# mode_color
mode_color 6 1 3

# aux
aux 0 0 0 1800 2000
aux 1 12 1 1000 1200
aux 2 14 1 1000 1200
aux 3 14 1 1800 2000

# adjrange

# rxrange

# rxfail

# master
set min_check = 1000
set max_aux_channels = 2
set min_throttle = 1100
set max_throttle = 1950
set use_unsynced_pwm = ON
set motor_pwm_protocol = MULTISHOT
set motor_pwm_rate = 8000
set serialrx_provider = SUMD
set vbat_scale = 112
set align_board_yaw = 90
set gyro_notch1_hz = 0
set gyro_notch2_hz = 160
set acc_hardware = NONE
set pid_process_denom = 1
set blackbox_rate_denom = 8
set ledstrip_visual_beeper = ON

# profile
profile 0

set dterm_notch_hz = 250
set dterm_notch_cutoff = 200
set pid_at_min_throttle = OFF
set setpoint_relax_ratio = 40
rateprofile 0

# rateprofile
rateprofile 0

set rc_rate = 108
set rc_rate_yaw = 108
set roll_srate = 80
set pitch_srate = 80

# 
```

martinbudden, 2016-12-10:
Fixed by PR https://github.com/betaflight/betaflight/pull/1790

martinbudden, 2016-12-10:
@TheAngularity , can you re-check, just to be sure?

closed 2016-12-10 by martinbudden

TheAngularity, 2016-12-10:
yep ... bug is fixed now 👍