betaflight/betaflight #1792: Failsafe problems with CC3D

rvzanten, 2016-12-10:
I searched the internet, but i could not seem to find a solution to fix my failsafe.
It seems the failsafe is getting triggerd when I turn off my tx, but the motors just keep spinning instead of that they go off. I hope somebody could point out the mistake i made or maybe the flaw in the code that causes this.

The betaflight dump:

`
# dump
# BetaFlight/CC3D_OPBL 3.0.1 Oct 18 2016 / 10:04:25 (48b7b4f)
name -
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 90 90 100 -1
servo 1 1000 2000 1500 90 90 100 -1
servo 2 1000 2000 1500 90 90 100 -1
servo 3 1000 2000 1500 90 90 100 -1
servo 4 1000 2000 1500 90 90 100 -1
servo 5 1000 2000 1500 90 90 100 -1
servo 6 1000 2000 1500 90 90 100 -1
servo 7 1000 2000 1500 90 90 100 -1

# servo mix
smix reset

feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -OSD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature RX_PPM
feature VBAT
feature MOTOR_STOP
feature FAILSAFE
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
beeper DISARM_REPEAT
beeper ARMED
beeper SYSTEM_INIT
beeper ON_USB
map AETR1234
serial 20 1 115200 57600 0 115200
serial 0 1 115200 57600 0 115200
serial 2 0 57600 57600 0 115200
aux 0 0 0 1400 2100
aux 1 1 0 1400 1725
aux 2 0 0 900 900
aux 3 0 0 900 900
aux 4 0 0 900 900
aux 5 0 0 900 900
aux 6 0 0 900 900
aux 7 0 0 900 900
aux 8 0 0 900 900
aux 9 0 0 900 900
aux 10 0 0 900 900
aux 11 0 0 900 900
aux 12 0 0 900 900
aux 13 0 0 900 900
aux 14 0 0 900 900
aux 15 0 0 900 900
aux 16 0 0 900 900
aux 17 0 0 900 900
aux 18 0 0 900 900
aux 19 0 0 900 900
adjrange 0 0 0 900 900 0 0
adjrange 1 0 0 900 900 0 0
adjrange 2 0 0 900 900 0 0
adjrange 3 0 0 900 900 0 0
adjrange 4 0 0 900 900 0 0
adjrange 5 0 0 900 900 0 0
adjrange 6 0 0 900 900 0 0
adjrange 7 0 0 900 900 0 0
adjrange 8 0 0 900 900 0 0
adjrange 9 0 0 900 900 0 0
adjrange 10 0 0 900 900 0 0
adjrange 11 0 0 900 900 0 0
adjrange 12 0 0 900 900 0 0
adjrange 13 0 0 900 900 0 0
adjrange 14 0 0 900 900 0 0
rxrange 0 1000 2000
rxrange 1 1000 2000
rxrange 2 1000 2000
rxrange 3 1000 2000
rxfail 0 a
rxfail 1 a
rxfail 2 a
rxfail 3 a
rxfail 4 h
rxfail 5 h
rxfail 6 h
rxfail 7 h
rxfail 8 h
rxfail 9 h
rxfail 10 h
rxfail 11 h
rxfail 12 h
rxfail 13 h
rxfail 14 h
rxfail 15 h
rxfail 16 h
rxfail 17 h
set mid_rc = 1543
set min_check = 1140
set max_check = 1900
set rssi_channel = 0
set rssi_scale = 30
set rc_interpolation = AUTO
set rc_interpolation_interval = 19
set rssi_ppm_invert = OFF
set input_filtering_mode = OFF
set roll_yaw_cam_mix_degrees = 0
set max_aux_channels = 6
set debug_mode = NONE
set min_throttle = 1120
set max_throttle = 1905
set min_command = 1000
set servo_center_pulse = 1500
set max_esc_throttle_jump = 0
set 3d_deadband_low = 1406
set 3d_deadband_high = 1514
set 3d_neutral = 1460
set 3d_deadband_throttle = 50
set enable_buzzer_p6 = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = OFF
set motor_pwm_rate = 400
set servo_pwm_rate = 50
set disarm_kill_switch = ON
set gyro_cal_on_first_arm = OFF
set auto_disarm_delay = 5
set small_angle = 25
set fixedwing_althold_dir = 1
set reboot_character = 82
set serialrx_provider = SPEK1024
set sbus_inversion = ON
set spektrum_sat_bind = 0
set spektrum_sat_bind_autoreset = 1
set telemetry_switch = OFF
set telemetry_inversion = OFF
set frsky_default_lattitude =  0.000
set frsky_default_longitude =  0.000
set frsky_coordinates_format = 0
set frsky_unit = IMPERIAL
set frsky_vfas_precision = 0
set frsky_vfas_cell_voltage = OFF
set hott_alarm_sound_interval = 5
set battery_capacity = 0
set vbat_scale = 100
set vbat_max_cell_voltage = 43
set vbat_min_cell_voltage = 33
set vbat_warning_cell_voltage = 35
set vbat_hysteresis = 1
set current_meter_scale = 400
set current_meter_offset = 0
set multiwii_current_meter_output = OFF
set current_meter_type = ADC
set align_gyro = DEFAULT
set align_acc = DEFAULT
set align_mag = CW90
set align_board_roll = 0
set align_board_pitch = 0
set align_board_yaw = 90
set max_angle_inclination = 700
set gyro_lpf = OFF
set gyro_sync_denom = 8
set gyro_lowpass_type = PT1
set gyro_lowpass = 90
set gyro_notch1_hz = 400
set gyro_notch1_cutoff = 300
set gyro_notch2_hz = 200
set gyro_notch2_cutoff = 100
set moron_threshold = 32
set imu_dcm_kp = 2500
set imu_dcm_ki = 0
set alt_hold_deadband = 40
set alt_hold_fast_change = ON
set deadband = 0
set yaw_deadband = 0
set throttle_correction_value = 0
set throttle_correction_angle = 800
set yaw_control_direction = 1
set yaw_motor_direction = 1
set tri_unarmed_servo = ON
set servo_lowpass_freq = 400
set servo_lowpass_enable = OFF
set airmode_activate_throttle = 1350
set failsafe_delay = 10
set failsafe_off_delay = 10
set failsafe_throttle = 1100
set failsafe_kill_switch = OFF
set failsafe_throttle_low_delay = 100
set failsafe_procedure = DROP
set rx_min_usec = 885
set rx_max_usec = 2115
set gimbal_mode = NORMAL
set acc_hardware = AUTO
set acc_lpf_hz =  10.000
set accxy_deadband = 40
set accz_deadband = 40
set acc_unarmedcal = ON
set acc_trim_pitch = 0
set acc_trim_roll = 0
set pid_process_denom = 1
set blackbox_rate_num = 1
set blackbox_rate_denom = 1
set blackbox_device = SERIAL
set blackbox_on_motor_test = OFF
profile 0

set pid_delta_method = MEASUREMENT
set yaw_p_limit = 500
set pidsum_limit = 700
set dterm_lowpass_type = BIQUAD
set dterm_lowpass = 100
set dterm_notch_hz = 260
set dterm_notch_cutoff = 160
set vbat_pid_compensation = OFF
set pid_at_min_throttle = ON
set iterm_throttle_gain = 0
set setpoint_relax_ratio = 30
set dterm_setpoint_weight = 200
set yaw_rate_accel_limit = 220
set rate_accel_limit = 0
set accum_threshold = 130
set yaw_accum_threshold = 32
set yaw_lowpass = 0
set p_pitch = 58
set i_pitch = 50
set d_pitch = 22
set p_roll = 43
set i_roll = 40
set d_roll = 20
set p_yaw = 70
set i_yaw = 45
set d_yaw = 20
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 50
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75
set level_sensitivity =  2.000
rateprofile 0
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 0
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 10
set tpa_breakpoint = 1650
`

andre92, 2016-12-11:
Your ppm receiver is set to hold channels or specific values. Read the manual for your receiver and set it to no pulses on failsafe

rvzanten, 2016-12-11:
Well its funny, if i test it out and turn my tx off when my motors are spinning it looks like Failsafe is triggered. I see the red failsafe icon in the right top of betaflight. Though, i think stage two is not triggered somehow. 

I mean, what does this red failsafe icon mean then if its not the failsafe being triggered which only happens if i turn off my tx?

andre92, 2016-12-12:
are your RC channels really centered around 1543? I never had to change the following:
set mid_rc = 1543 (1500)
set min_check = 1140 (1100)
set failsafe_throttle = 1100 (1000)
min and max throttle are unrelated to mid_rc, min_check and max_check
min_throttle should be set to where the motors start plus a little margin, max throttle can be left at 2000. (unless motors max out way before that) Are your ESCs calibrated? These values look a bit unusual. 
set min_throttle = 1120 (1070)
set max_throttle = 1905 (2000)

rvzanten, 2016-12-12:
> are your RC channels really centered around 1543? I never had to change the following:

Yes, they are. Except the throttle.

> Are your ESCs calibrated? 

I don´t believe my ESCs are calibrated. But I never did when i was still working with Librepilot and then it all worked fine. 

Iam not sure where to look for now since failsafe does actually get triggered, only Stage 2 doesnt do anything. The motors just keeps spinning. And like you said, failsafe_throttle is set to 1100 which is much lower then my minimum throttle and stops my motors when i test it out in the betaflight motors tab. I have no clue where to look. :(

andre92, 2016-12-14:
Librepilot doesn't calibrate the ESCs but calibrates the flight controller's output for each ESC if I remember correctly. If your ESCs are running BLHeli or SimonK, you could try calibrating them, and then setting Betaflight back to the defaults. Your failsafe is set to DROP, which should disarm the craft. Could you confirm the craft goes from armed to disarmed state when failsafe activates?

borisbstyle, 2016-12-19:
Please set the failsafe manually if you use PPM.

This manual will give you a good clue
http://fpvobsession.com/setting-up-failsafe-on-frsky-beebrain/

closed 2016-12-19 by borisbstyle