betaflight/betaflight #1793: OSD in Configurator

QuadroBro, 2016-12-10:
Hey guys, the OSD tab in the configurator on the last few builds is unuseable. It cuts off and there's no way to interact with it or choose settings. Seen this on multiple omnibus variants.

martinbudden, 2016-12-11:
Have you made sure you have the latest version of the 3.1 build and the latest version of the configurator?

QuadroBro, 2016-12-11:
I refreshed the repo prior to downloading and the configurator auto updates.

QuadroBro, 2016-12-11:
Here's what I'm looking at. About to flash 806.

[image]

QuadroBro, 2016-12-11:
No change with 806

[image]

jflyper, 2016-12-11:
Did you "reload" in your extensions manager?

rafl, 2016-12-11:
This seems to be caused by the addition of `OSD_{PITCH,ROLL,YAW}_PIDS` to the OSD, which isn't yet supported in version 1.8.4 of the configurator. The latest git version of the configurator appears to work just fine.

I don't think this is a betaflight issue, but an issue with the configurator behaving very oddly when talking to a flight controller that supports more OSD elements than are known to the configurator.

I've observed the same issue with https://github.com/betaflight/betaflight/pull/1811.

mikeller, 2016-12-11:
@rafl is correct there. In general, when using unreleased builds of the firmware, you should be using latest of the configurator from the repository as well.

closed 2016-12-19 by borisbstyle

mactac, 2017-07-26:
I am using the most current version, and I am seeing the same problem.

jflyper, 2017-07-26:
@mactac The thread is closed, and issue was resolved. If you really have a problem, it can be fixed by using the up to date dev version of the configurator. If not, please submit a new issue.