betaflight/betaflight #1794: Mavlink telemetry broken on CC3D

bluehallu, 2016-12-10:
If MSP and Mavlink telemetry are selected in one port (UART1 in my case), saving and rebooting hangs the board (escs beep indefinitely) and upon restart all settings in the ports tab are reverted to default (even for other ports).

If only telemetry is selected, settings do save successfully but OSD isn't receiving anything.

Betaflight 3.0.1
Configurator 1.8.4
Falcon 250 customised CC3D
Falcon 250 customised OSD (Mavlink v1 as far as I can tell)

mikeller, 2016-12-13:
Yup, that's a bug in 3.0.1. It has been fixed in 3.1.

closed 2016-12-19 by borisbstyle

bluehallu, 2016-12-20:
@borisbstyle @mikeller Is [this PR](https://github.com/betaflight/betaflight/pull/826) what fixes it? I've tested with no success and it does say on the PR itself it's disabled for CC3D due to dataflash <128k, which is weird given that it works out of the box with OpenPilot?

mikeller, 2016-12-20:
@bluehallu The PR that I was referring to is #1622. This fixes sharing of a port for Mavlink telemetry in general, for all targets that support Mavlink. But if CC3D doesn't support Mavlink, then this won't fix it obviously.

Betaflight has a lot of advanced flight performance related features that OpenPilot doesn't have. As a drawback, it pushes the limits of the available flash space on F1 based targets, and compromises have to be made with respect to supporting other features.