betaflight/betaflight #1796: Adding max gps speed in osd stats

yks-aircrafts, 2016-12-11:
Hello !
I use a omnibus f3 pro with gps just to show the speed of the drone.
Is there a way to show max speed in stats at the end of the run ?

jflyper, 2016-12-11:
As I see the current code, there is MAX SPEED entry in the stats screen albeit it is only displayed if the vehicle has a GPS fix at the time of the display.

yks-aircrafts, 2016-12-12:
Many thanks for your answer !
My mistake I had just no gps Fix in 3.0.1, it displays the speed and the max speed in the stats, but with no unit (no kph in metric neither mph in imperial).
I tried to flash my board with 3.1 to test it but, it's impossible to set gps mode to ublox. After save and reboot it stays on NEMA mode.
But I'll wait the official release of 3.1.

jflyper, 2016-12-12:
@yks-aircrafts 
If you are having problem with the ublox, you should submit a separate issue, because the problem does not get fixed automagically; has to have someone looking into it ;)

closed 2016-12-19 by borisbstyle