betaflight/betaflight #1808: One channel not responsive

demersons, 2016-12-11:
Has anyone had a problem with FC not sending correct signal to one motor?   We have replaced ESC's FC and now think its the Firmware flash is wrong.    Does this make sense to anyone?

SurgeonFPV, 2016-12-11:
Have you tried reflashing?

demersons, 2016-12-11:
Hey thanks for getting back to us. Yes we've tried re-flashing we tried re-flashing with beta flight with clean flight and we replace the FC.  I'm thinking the reflash isn't taking. We have a fc on order.  Don't know what else to do.  
Dan

SurgeonFPV, 2016-12-11:
If you have tried a new FC, then maybe its not that. Maybe its the motor? Tried switching a different motor to the output on the fc?

demersons, 2016-12-11:
Good idea....we effectively did that.   Moved the white and black wire to another ESC/Motor and the problem followed.   Do you think Beta flight could be sending the problem to the FC? 
 Dan Salter 
619 823 1573

borisbstyle, 2016-12-19:
This sounds like a hardware issue. Please test on a new FC.

Also next time provide more information about the target, motor etc...

closed 2016-12-19 by borisbstyle