betaflight/betaflight #1810: Yaw Gyro constant noise

SurgeonFPV, 2016-12-11:
Hi All.
I'm hoping someone can help me with my F4 yaw gyro issues. I began to notice on my one quad that i would see some yaw vibrations when flying forward on tri blades. I've adjusted Pids, and while its still flyable, its annoying and is much more pronounced against the wind. When i put on dual blades for some testing, my quad is impossible to fly, the yaw vibrations are so bad my camera goes blurry. I've done some bbox and can see the yaw gyro has constant noise. 

I've tried stock pids and even lowering my yaw down from 70 to about 50. And its still the same.
I tried another F4 quad, same issue. Both the same board with the 6000 gyro 
Both soft mounted with o-rings
Both different frames, but similar shape.
Both different motors (in case one set is noisier than the other)
I then tried my x-racer F3 quad, and my yaw gyro is quiet.

My FC is very well soft mounted, yet im still getting noise.
Has anybody experienced this or can someone tell me what i need to do?

The F4 i have is the pin header version of the Flip32 F4 as at readytoflyquads, however these are from aliexpress so it may be a clone or just not labelled.

Thanks in advance

Martin

mikeller, 2016-12-11:
Hi Martin.

I think setting up notch filtering should help you with your troubles: https://www.youtube.com/watch?v=iUASxYf-6sg

Cheers

SurgeonFPV, 2016-12-12:
Thanks, ill give that ago. see how i go

martinbudden, 2016-12-20:
Closing since no further response from original author.

closed 2016-12-20 by martinbudden

SurgeonFPV, 2016-12-20:
Hi guys, Sorry, i just resolved this tonight, it wasnt notch filters, my yaw noise was between the 0-100hz range. I disconnected my vtx and cam and did the same test, no noise. Looks like i have a ground loop. Rewired everything and its ok now.