betaflight/betaflight #1817: USE_UART1_TX_DMA in common.h considered not suitable for some targets

jflyper, 2016-12-13:
I believe this option is defined with use case of UART1 as blackbox output in mind,  but...

- It lacks flexibility of configuration.
- Sometimes meaningless; targets such as AIRBOT F4 and OMNIBUS F4 series designates UART1 as SBUS, in which case the TX DMA is completely nonsense.

jflyper, 2017-01-08:
As all UART DMAs are disabled by default to avoid possible conflict with Dshot, this issue is now pointless.

I renamed it to remind us that we may have a chance to revive them for non-conflicting targets.

hydra, 2019-07-14:
should we close this or update it?

mikeller, 2019-07-14:
This is potentially one to flow into the overhaul of how UART configuration is done (general idea: set options by UART, not by UART function as it is now), so we can keep it around. Not sure when the overhaul is going to happen, @blckmn said at some point he was going to look into it.

blckmn, 2019-07-14:
Yes, I’m looking into it. The precursor is getting the param groups settable by index. Not far now.

closed 2026-09-26 by blckmn