betaflight/betaflight #1830: Software BIND pin output for RX

fishpepper, 2016-12-14:
I was wondering if adding a software bind output would be a nice thing to have.

I am currently building a tiny f3 fc that includes a frsky compatible rx ([tinyfish FC](http://fishpepper.de/projects/tinyfish/) ). I would like to have the option to set/enter the bind mode (and probably bootloader mode) for the rx from within betaflight.
I am thinking of something similar that was done for the spektrum satellites. But configurable and maybe having 2 outputs would be nice (to enter bootloader mode on the rx to allow firmware upgrades).
There are probably other fcs with integrated rx coming soon so this might be useful to others as well.

Is there interest in this? Then i might code this part and do a pull request ;)

mikeller, 2016-12-14:
Software BIND / bootloader mode is not quite as nice as FrSky with an SPI attached CC2500, but it would still make a nice and welcome addition. :-P

There's already a bunch of small RX integrated FC boards out there:

http://www.banggood.com/Beecore-F3_EVO_Brushed-ACRO-Flight-Control-Board-DSM2-For-Inductrix-Tiny-Whoop-Eachine-E010-p-1089021.html
http://www.banggood.com/Eachine-AIOF3_BRUSHED-Flight-Control-Board-Built-in-OSD-FrskyFlyskyDSM2DSMX-Receiver-Betaflight-p-1107427.html

They presumably all use the 'midelic' RX firmware running on an Atmega MCU, so the steps required to get them into the boot loader (they seem to be wired for BIND on a button) should be reasonably similar so if this could be made flexible enough to accommodate these it would be great.

fishpepper, 2016-12-14:
I know these boards, they just add unnecessary complexity using the atmega. The cc2510 incorporates a nice and powerful 8051 cpu. I also wrote a bootloader for those, so it is possible to upgrade the firmware from within betaflight.

ps2017, 2017-02-26:
I just bought a FPV drone from Rotorgeeks fully built with stock tune and unable to bind, they performed the bind using a separate RX to the satellite connected to the board and informed me that they were unable to bind using CLI because of a problem with the latest version of Betaflight and not a problem with the hardware. When can I expect this software problem be fixed?

mikeller, 2017-02-26:
@ps2017: Your setup is entirely different (Spektrum Satellite) to what is discussed here (FrSky). Spektrum satellite binding is supported for most boards, except for some F4 based boards where an external inverter makes it impossible. Without knowing what board it is you've got I can't give you more information, so if you are interested, please open a new issue.

ps2017, 2017-02-26:
Hello and apologies for the frsky posting. The controller is a Rotorgeeks SSD F3 control board, here is a link to it which provides additional tech info:
http://rotorgeeks.com/index.php?route=product/product&product_id=676

ps2017, 2017-02-26:
Hello and apologies for the frsky posting. The controller is a Rotorgeeks SSD F3 control board, here is a link to it which provides additional tech info:
http://rotorgeeks.com/index.php?route=product/product&product_id=676

[http://rotorgeeks.com/image/cache/data/electronics/FC/RG-SSD-top.800-800x600.jpg]<http://rotorgeeks.com/index.php?route=product/product&product_id=676>

Rotorgeeks SSD F3 control board<http://rotorgeeks.com/index.php?route=product/product&product_id=676>
rotorgeeks.com
Special introductory pricing! The Rotorgeeks SSD is an F3 based flight controller with best-in-class components, an intelligent layout and a great price.

If you still need me to create a new case please provide instructions on how to do so as its not apparent from the cases web page, thank you.

mikeller, 2017-02-27:
@ps2017: What port is your Spektrum satellite Rx connected to? Has to be UART3 for Spektrum bind to work on this board.

ps2017, 2017-02-27:
Its on UART3, find attached, just odd the CLI didn't allow bind so RG had to hook it upto a separate Rx connected to the SAT

Do you also have optimal PID's for this?

closed 2018-07-07 by mikeller