betaflight/betaflight #1834: BF3.1 Airbot FW Buzzer issue

Oriolkun, 2016-12-15:
Hello, I have an issue using BF3.1 target AIRBOT (F4 Revo with VBat anb Buzz from Banggood)
The buzzer/beeper connected to Buzz pins does not work. Firmware target REVO makes the the buzzer beeping all the time. Like other issues reported. 
Then I changed the firm to the AIRBOT BF3.1 but it does not beep. only when connecting supply/USB an instant super short sound and no sound again. 
Do you know if there is any way to use firm 3.1 with Buzzer? Maybe using motor 6...? Thanks.

atomiclama, 2016-12-15:
You should be able to use the resource command to map to any free output.
I'll look into this specific point.

Oriolkun, 2016-12-15:
Hi, there is any way to check what I/O (A00, A01...B01) correponds to a phisical pinout? any picture of that AIRBOT F4 with the pinout named as (A00, A01 ... etc) ? Thanks

AndersHoglund, 2016-12-15:
We have had other reports, telling that when changing to target AIRBOTF4 firmware the beeper start working fine. I can not see what has changed since then.... Ref #1736 #1776 etc etc 

To check connection between CPU pins and externalconnectors/pads you need the schematics from your vendor/manufacturer. Or reverse engineer it with a DVM.

AndersHoglund, 2016-12-15:
If you want beeper output on motor 6 then do this re-mapping:

  resource BEEPER A00
  resource MOTOR 6 NONE

But the should not be needed, and it would also require a buffered beeper.

atomiclama, 2016-12-15:
Maybe some funnies with this banggood board again.

AndersHoglund, 2016-12-15:
Yupp, new rev maybe? Early on I suggested they should have used a free pin for a dedicated beeper output, not the LED1 pin. And also connect Spektrum sat to UART3 Rx not UART1. But I can not find any info that this has been done yet. 
More noise here: https://www.rcgroups.com/forums/showthread.php?2731917-Airbot-F4-Flip32-F4-Thread

jflyper, 2016-12-15:
@atomiclama 
Do some trace analysis with your DMM.
Do you know what to do?

atomiclama, 2016-12-15:
Yeah I know what to do ;-)
Everything I could wish for at work.
I'll have a look at lunch time

jflyper, 2016-12-15:
Cool. The best & shortest path to the working beeper. 👍

Oriolkun, 2016-12-15:
@AndersHoglund As I can see in the other posts they say the continuous beeping stops but I don't kow how they solved the problem. These posts are closed by now and I don't know how to ask them if they really solved and make funcional buzzer in buzzer pins on Revo F4 board with BF3.1 AIRBOT firm target.
So maybe I can invoke them to ask about how did they solved this issue... @puaka and @HMFHOBBY
Thanks

Oriolkun, 2016-12-15:
@AndersHoglund Do you know why the CLI commands are different in that configurations of using Motor6 than yours? 

> Betaflight 3.1 update: No special version needed, just these commands in cli:
> beeper_od = ON
> beeper_inverted = OFF
> resource BEEPER A00

AndersHoglund, 2016-12-15:
That was with an earlier version of the REVO target. Now defaults are OK:

    # get beep
    beeper_inversion = OFF
    Allowed values: OFF, ON
    
    beeper_od = ON
    Allowed values: OFF, ON

Also. @puaka said: "Awesome! It is fine now, everything works as expected. So revo F4 from banggod which has buzzer pad need to use AIRBOTF4 target."
I can only interpret that in one way. But you may hopefully get an answer directly.

Oriolkun, 2016-12-15:
Yes, @puaka should have a working revo F4 with AIRBOTF4 as me. But if he had working buzzer might be because have some difference on Beeper resource remapping. If he chanched any value or not use the default source they might solved the problem. I can say that in my case, de default source does not work with buzzer. I've looking for some extra sourcers to apply de remapping and I have to try some (A12, A11 or A10)

> resource BEEPER **##**

AndersHoglund, 2016-12-15:
For all I know, AIRBOTF4 had the beeper connected to PB04, which is LED1 on REVO. 
And these are the default resource  assignments in current firmware... 

    #if defined(AIRBOTF4)
    #define BEEPER                  PB4
    #define BEEPER_INVERTED

You can check this yourself with the "resource" command: 

    # version
    # Betaflight/AIRBOTF4 3.1.0 Dec 15 2016 / 13:08:30 (d93224b)
    
    # resource
    resource BEEPER B04

Nb, I don't have a AIRBOTF4 to actually try it, this is with the AIRBOTF4 firmware flashed on my REVO.

jflyper, 2016-12-15:
I confirm the beeper on B04 on OMNIBUSF4 (AIRBOTF4 + OSD).

atomiclama, 2016-12-15:
OK just wired up my beeper on this board
http://www.banggood.com/Raceflight-Betaflight-CC3D-REVO-F4-STM32F405-Flight-Controller-with-VbatBuzzer-p-1087561.html

Working fine on version
` Betaflight/AIRBOTF4 3.1.0 Dec 14 2016 / 07:36:21 (a282d72)`

Tested by issuing play_sound and normal operation.

Same IO mapping as above
'resource BEEPER B04'

atomiclama, 2016-12-15:
@Oriolkun did you perform a default after switching from REVO to AIRBOTF4 builds?

Oriolkun, 2016-12-15:
@atomiclama Perform a default? sorry, I don't understand what you mean. I have a Revo F4 of those and the last thing i did were flashing Betaflight/AIRBOTF4 3.1.0 as you say with Betaflight. I haven't touched anything more. Only pluged the buzzer on Bzz pins + and - as always and it does not sound, only a short-beep at first when plug USB. The other stuff like sbus and motors works well. I will check what you say about the resource BEEPER B04 if i  have the same.

There is a way to perform a default settings?

atomiclama, 2016-12-15:
on the CLI
default or defaults

puaka, 2016-12-15:
@Oriolkun no CLI is needed, flashed with 3.1 Airbot and boom... everything solved. No idea why yours not working.

puaka, 2016-12-15:
[image]

the firmware version that works, i did not see any changes remarked for Airbot target, so later versions should work.

AndersHoglund, 2016-12-15:
If the FC was flashed with REVO firmware, and then changed to AIRBOTF4, it is absolutely vital to do a "Reset Settings" via GUI, or "defaults" via CLI. And then also a power cycle.
Othervise the REVO resource mapping might still be in use.

blckmn, 2016-12-15:
@andershoglund you make a good point. I'll add a guard against that happening. The config should now only be valid for the target it was created for.

Oriolkun, 2016-12-15:
@AnderHoglund @puaka @atomiclama Thanks!!
Finally!!! it was all about donig a "defaults" via CLI, then restart and that beep started that sound like the music of the paradise! Now the buzzer works perfect on a Revo F4 with Airbot BT3.1. I have tested to make it sound by the controller and works as expected.

I didn't know after flashing is necessary to make a manual "defaults" to reset the old parameteres/resources...
Thanks for all your helps.

closed 2016-12-15 by Oriolkun

AndersHoglund, 2016-12-15:
Good. 
It was not the flashing in it self that required a default reset. It was the change of target firmware to a another type with other default values. Those did not get set automatically, the old ones remained. Needed a manual reset to take effect.

povlhp, 2016-12-16:
Have the BangGood Revo F4. Installed BF 3.1 from Jenkins. The AirbotF4.hex 
With a multimeter and beeper active voltage jumps all around. Using Hz mode it seems how have a frequency of around 3.8kHz when beeper on. Next step here is to solder a beeper.