betaflight/betaflight #1838: ESC can not be calibrated (not entering in programming mode)

Steadyspike, 2016-12-15:
Hi,

I can not calibrate my 4xESC using Betaflight using the proper workflow (click motor tab, max throttle slider, connect power to ESC and min throttle slider). It happens because I can not enter in programming mode (there's no beep in max/min slider position and I check in BLHelusuite the PPM Min/Max values are not modifyed).

I guess the problem is in the signal generated by FC. In min throttle, it's around 900us and in max throttle is 1800us, so it doesn't arrive to 2000us and that's why the ESC doesn't enter in programming mode. I have same problem using other settings (PWM/oneshot125, different firmware FC releases...) and also using BLHeli suite motor tab. Find attached osciloscope ESC signal capture - min/max throttle using Betaflight throttle slider - (time scale is 500us/div):
[image]
[image]

I have connected a servo tester directly to each ESC/motor and calibrated it correctly (signal generated by servo tester was 900-2075us). So it seems that the signal generated by the FC doesn't allow entering in programming mode...

My settings are:
Betaflight: 1.8.4
SP F3 EVO: 3.01, Minimum command: 1000, Maximum throttle: 2000
ESC: DYS XS30A, Rev 16.5, programming by Tx enable, PPM min throttle: 1040, PPM max throttle1832

Any idea/suggestion?

mikeller, 2016-12-15:
Just a stab in the dark, but could be a faulty crystal on your FC. You wouldn't happen to have another FC that you could try with?

Steadyspike, 2016-12-15:
It's a pitty but I have no additional FC. I was wondering whether it has sense purchase another SP F3 EVO but before I would like to assure it's not a bad setting, bas soldering, SPF3EVO & DYS incompatibility...

AndersHoglund, 2016-12-16:
Since you have a scope connected you could probably try to compensate by adjusting min_command and max_throttle so you get the desired 1000 .... 2000 range as measured on the pins, then retry a calibration. Just a test, if the board is faulty it should be replaced anyhow. There would probably turn up other issues as well.

Steadyspike, 2016-12-16:
The previous screenshot is done with Max throttle=2000. Betaflight doesn't allow a value higher than 2000 in max throttle. So max I can get is 1800us (instead of the 2000us required). Does someone know when ESC enters in program mode? I have read in all documentation that ESC enters in programming mode once it's in "max throttle". What does it mean? Can I configure a value in the ESC that triggers it or it's harcoded? 

I just want to assure the board is faulty. Perhaps some one could provide me a test point so that I could confirm clock is not working properly...

Steadyspike, 2016-12-16:
I have read in a foro (http://electronics.stackexchange.com/questions/164803/whats-the-relationship-between-esc-pwm-input-and-output) that: "... The ESC is put into calibration mode by starting at full throttle (>1.5ms), then going to low throttle (<1.5ms) after a few seconds....".

So does it mean that it should work with the 1.8ms PWM signal generated by FC? it sounds me strange becasue I configure Maximum throttle: 2000 in Betaflight...

borisbstyle, 2016-12-19:
@Steadyspike 
You do understand that oneshot does not use signals of 1.8ms? You have to use normal PWM in order to get that

closed 2016-12-19 by borisbstyle