betaflight/betaflight #1842: Align MAG sensor on a vertical plane - add more positions to align_mag

nhadrian, 2016-12-17:
Hi,

I have a race quad with closed hull where I have limited free space inside.
The only possibility of adding a mag sensor BOB is to align it vertically inside the frame:
[image]
But it looks like the possible mag alignments (align_mag cli parameter) doesn't include any vertical planes.

If possible, please add some option for that.

Regards
Adrian

borisbstyle, 2016-12-20:
I believe that the chip itself does not support any other MAG aligns. This is not a part of FC code. The alignement is being passed to the MAG sensor itself.

closed 2016-12-20 by borisbstyle

nhadrian, 2016-12-20:
Thanks for your reply. If I know right, these are 3D mag sensors, so they'll provide a 3D mag vector as a measurement result.

The mag orientation is defined in the `alignSensors(int32_t *dest, uint8_t rotation)`, here:
```
switch (rotation) {
    default:
    case CW0_DEG:
        dest[X] = x;
        dest[Y] = y;
        dest[Z] = z;
        break;
    case CW90_DEG:
        dest[X] = y;
        dest[Y] = -x;
        dest[Z] = z;
        break;
    case CW180_DEG:
        dest[X] = -x;
        dest[Y] = -y;
        dest[Z] = z;
        break;
    case CW270_DEG:
        dest[X] = -y;
        dest[Y] = x;
        dest[Z] = z;
        break;
    case CW0_DEG_FLIP:
        dest[X] = -x;
        dest[Y] = y;
        dest[Z] = -z;
        break;
    case CW90_DEG_FLIP:
        dest[X] = y;
        dest[Y] = x;
        dest[Z] = -z;
        break;
    case CW180_DEG_FLIP:
        dest[X] = x;
        dest[Y] = -y;
        dest[Z] = -z;
        break;
    case CW270_DEG_FLIP:
        dest[X] = -y;
        dest[Y] = -x;
        dest[Z] = -z;
        break;
    }
```

Since X and Y are changed in some cases above, why Z and X and Y can't be changed alltogether?
For example, if sensor flipped forward by 90 degrees and then rotated around yaw by 90 degrees, it would look like:
```
case VERTICAL_0_DEG:
        dest[X] = z;
        dest[Y] = y;
        dest[Z] = -x;
        break;
    }
case VERTICAL_CW90_DEG:
        dest[X] = y;
        dest[Y] = -z;
        dest[Z] = -x;
        break
   }
case VERTICAL_CW180_DEG:
        dest[X] = -z;
        dest[Y] = -y;
        dest[Z] = -x;
        break
   }
case VERTICAL_CW270_DEG:
        dest[X] = -y;
        dest[Y] = -z;
        dest[Z] = -x;
        break
   }
```

Regards
Adrian