betaflight/betaflight #1844: 827 build. Can't save PIDs. FC restarts.

GrishaLutsenko, 2016-12-17:
So that I found after flashing of 827 build. I can change PIDs from RC, but can't from GUI. When I press save FC makes restart and CDC hangs. So I have to power cycle FC.

GrishaLutsenko, 2016-12-17:
Tried 805, all works

SandroGrzicic, 2016-12-17:
Can confirm, latest BF master that I just flashed causes the configurator / FC to stop responding once I hit Save under PID tuning.

DieHertz, 2016-12-17:
Communication freezes after sending `MSP_SET_FILTER_CONFIG`, investigating.

AndersHoglund, 2016-12-17:
Confirmed. build 823 OK, build 824 broken. Four commits in there.....

AndersHoglund, 2016-12-17:
Commit 13a7de8d3b30a4fc78cb7822805dad4be41fcd11 OK, 
next commit 08a550140bc131b5f80aecd4187a8188ac832bf4 broken.
I.e PR #1771 brakes the build.

DieHertz, 2016-12-17:
```
commit 17494840a546ea73577d2701f4cea2868a05447e
Author: Martin Budden <mjbudden@gmail.com>
Date:   Tue Dec 13 10:00:43 2016 +0000

    Moved gyro and acc detection code into init functions```

It's this one

DieHertz, 2016-12-17:
Seems like it does not live with the fact that `MSP_SET_FILTER_CONFIG` has to reinitialize gyro.

AndersHoglund, 2016-12-17:
This is very strange, I can verify your bfindings, 17494840a546ea73577d2701f4cea2868a05447e failing. 
I also retested my findings and this version works:

    # version
    # Betaflight/SPRACINGF3 3.1.0 Dec 17 2016 / 23:21:39 (13a7de8)
while this fails:

    # version
    # Betaflight/SPRACINGF3 3.1.0 Dec 17 2016 / 23:30:07 (08a5501)

What is going on here? Do we have a problem that was partly fixed and then re-introduced again. You found the first occurrence while I found the last... Weird.

AndersHoglund, 2016-12-17:
Aha, not so strange after all. We have found the same thing. You found the original commit on the PR #1771 branch, while I found the merge commit that integrated #1771 on master. Same thing, at different times.

closed 2016-12-19 by borisbstyle