betaflight/betaflight #1845: ARF3 DSHOT  can not work

tianbin4279, 2016-12-17:
arf3 3.1 dshot can not work
According to f3evo re-made the target
but Motor2  not worl in dshot,multishot is ok
target.c:
`#include <stdint.h>

#include <platform.h>
#include "drivers/io.h"

#include "drivers/timer.h"
#include "drivers/timer_def.h"
#include "drivers/dma.h"

const timerHardware_t timerHardware[USABLE_TIMER_CHANNEL_COUNT] = {
    // PPM / UART2 RX
    DEF_TIM(TIM8,  CH1,  PA15, TIM_USE_PPM,   0 ),  // PPM
    DEF_TIM(TIM3,  CH4,  PB1,  TIM_USE_MOTOR, 1 ),  // PWM1
    DEF_TIM(TIM3,  CH2,  PA7,  TIM_USE_MOTOR, 1 ),  // PWM2
    DEF_TIM(TIM2,  CH3,  PA2,  TIM_USE_MOTOR, 1 ),  // PWM3 
    DEF_TIM(TIM2,  CH2,  PA1,  TIM_USE_MOTOR, 1 ),  // PWM4 
    DEF_TIM(TIM3,  CH3,  PB0,  TIM_USE_MOTOR, 1 ),  // PWM5
    DEF_TIM(TIM3,  CH1,  PA6,  TIM_USE_MOTOR, 1 ),  // PWM6
    DEF_TIM(TIM15, CH2,  PA3,  TIM_USE_MOTOR, 1 ),  // PWM7
    DEF_TIM(TIM2,  CH1,  PA0,  TIM_USE_MOTOR, 1 ),  // PWM8
    DEF_TIM(TIM2,  CH3,  PB10, TIM_USE_MOTOR, 1 ),  
    DEF_TIM(TIM2,  CH4,  PB11, TIM_USE_MOTOR, 1 ),  
    DEF_TIM(TIM1,  CH1,  PA8,  TIM_USE_LED | TIM_USE_TRANSPONDER, 1 ),  // LED_STRIP / TRANSPONDER
};`

target.h:
`
#pragma once

#define TARGET_BOARD_IDENTIFIER "ARF3"

#define CONFIG_FASTLOOP_PREFERRED_ACC ACC_DEFAULT

#define LED0                    PB8

#define BEEPER                  PC15
#define BEEPER_INVERTED

#define USE_EXTI
#define MPU_INT_EXTI            PC13
#define EXTI15_10_CALLBACK_HANDLER_COUNT 2 // MPU_INT, SDCardDetect
#define USE_MPU_DATA_READY_SIGNAL
#define ENSURE_MPU_DATA_READY_IS_LOW

#define USE_MAG_DATA_READY_SIGNAL
#define ENSURE_MAG_DATA_READY_IS_HIGH

#define USE_ESC_TELEMETRY

#define GYRO
#define USE_GYRO_SPI_MPU6500

#define ACC
#define USE_ACC_SPI_MPU6500

#define ACC_MPU6500_ALIGN       CW180_DEG
#define GYRO_MPU6500_ALIGN      CW180_DEG

#define BARO
#define USE_BARO_BMP280

#define MAG
#define USE_MAG_AK8963
//#define USE_MAG_HMC5883 // External

#define MAG_AK8963_ALIGN CW90_DEG_FLIP

//#define SONAR

#define USB_IO

#define USE_VCP
#define USE_UART1
#define USE_UART2
#define USE_UART3
#define SERIAL_PORT_COUNT       4

#define USE_ESCSERIAL
#define ESCSERIAL_TIMER_TX_HARDWARE 0 // PWM 1

#define UART1_TX_PIN            PA9
#define UART1_RX_PIN            PA10

#define UART2_TX_PIN            PA14 // PA14 / SWCLK
#define UART2_RX_PIN            PA15

#define UART3_TX_PIN            PB10 // PB10 (AF7)
#define UART3_RX_PIN            PB11 // PB11 (AF7)

#define USE_I2C
#define I2C_DEVICE              (I2CDEV_1) // PB6/SCL, PB7/SDA

#define USE_SPI
#define USE_SPI_DEVICE_1 // PB9,3,4,5 on AF5 SPI1 (MPU)
#define USE_SPI_DEVICE_2 // PB12,13,14,15 on AF5 SPI2 (SDCard)

#define SPI1_NSS_PIN            PB9
#define SPI1_SCK_PIN            PB3
#define SPI1_MISO_PIN           PB4
#define SPI1_MOSI_PIN           PB5

#define SPI2_NSS_PIN            PB12
#define SPI2_SCK_PIN            PB13
#define SPI2_MISO_PIN           PB14
#define SPI2_MOSI_PIN           PB15

#define USE_SDCARD
#define USE_SDCARD_SPI2

#define SDCARD_DETECT_INVERTED

#define SDCARD_DETECT_PIN                   PC14
#define SDCARD_SPI_INSTANCE                 SPI2
#define SDCARD_SPI_CS_PIN                   SPI2_NSS_PIN

// SPI2 is on the APB1 bus whose clock runs at 36MHz. Divide to under 400kHz for init:
#define SDCARD_SPI_INITIALIZATION_CLOCK_DIVIDER 128
// Divide to under 25MHz for normal operation:
#define SDCARD_SPI_FULL_SPEED_CLOCK_DIVIDER     2

// Note, this is the same DMA channel as UART1_RX. Luckily we don't use DMA for USART Rx.
#define SDCARD_DMA_CHANNEL_TX               DMA1_Channel5
#define SDCARD_DMA_CHANNEL_TX_COMPLETE_FLAG DMA1_FLAG_TC5

#define MPU6500_CS_PIN                   PB9
#define MPU6500_SPI_INSTANCE             SPI1

#define BOARD_HAS_VOLTAGE_DIVIDER
#define USE_ADC
#define ADC_INSTANCE            ADC2
#define VBAT_ADC_PIN            PA5
#define CURRENT_METER_ADC_PIN   PA4
#define RSSI_ADC_PIN            PB2

#define LED_STRIP

#define TRANSPONDER

#define ENABLE_BLACKBOX_LOGGING_ON_SDCARD_BY_DEFAULT

#define DEFAULT_RX_FEATURE      FEATURE_RX_PPM
#define DEFAULT_FEATURES        (FEATURE_TRANSPONDER | FEATURE_BLACKBOX | FEATURE_RSSI_ADC | FEATURE_CURRENT_METER | FEATURE_TELEMETRY)

#define SPEKTRUM_BIND
// USART3,
#define BIND_PIN                PB11

#define USE_SERIAL_4WAY_BLHELI_INTERFACE

// IO - stm32f303cc in 48pin package
#define TARGET_IO_PORTA         0xffff
#define TARGET_IO_PORTB         0xffff
#define TARGET_IO_PORTC         (BIT(13)|BIT(14)|BIT(15))
#define TARGET_IO_PORTF         (BIT(0)|BIT(1)|BIT(4))

#define USABLE_TIMER_CHANNEL_COUNT 12 // PPM, 8 PWM, UART3 RX/TX, LED Strip
#define USED_TIMERS             (TIM_N(1) | TIM_N(2) | TIM_N(3) | TIM_N(8) | TIM_N(15))
`

mikeller, 2016-12-19:
Hi @tianbin4279.

The reason for this is that motor 1 & 2 have a resource conflict (both use timer 3):

```
DEF_TIM(TIM3, CH4, PB1, TIM_USE_MOTOR, 1 ), // PWM1
DEF_TIM(TIM3, CH2, PA7, TIM_USE_MOTOR, 1 ), // PWM2
```

You might be able to get DShot to work by using motors 1, 3 & 7, and either PPM (PA15) or LED (PA8) for the 4th motor.

borisbstyle, 2016-12-21:
If this target is equal to f3 evo than this is a known limitation of this board.

You can try resource remap feature for that motor

closed 2016-12-21 by borisbstyle