betaflight/betaflight #1848: Serial Receiver: Adding SRXL Protocol

nepomukmaier, 2016-12-17:
Hello,

I would like to use a Multiplex Flexx receiver, which is working with the SRXL protocol. Is it possible to implement this protocol?

A description of the protocol can be found here (pages 3 & 4 are in English):
https://www.multiplex-rc.de/Downloads/Multiplex/Schnittstellenbeschreibungen/srxl-multiplex-v2.pdf

A more general description can be found here (Serial Receivers --> SRXL):
https://github.com/cleanflight/cleanflight/blob/master/docs/Rx.md

Maybe it would beneficial to include all mentioned SRXL versions.

nepomukmaier, 2016-12-21:
I have just checked the cleanflight code and the protocol is already a part of it, so I hope that an implementation in betaflight would not be too  complicated. 

I only have basic programming skill, but I am willing to support wherever I can.

atomiclama, 2016-12-21:
I'll have ago next year if no one else has picked it up. 
Although I have nothing to test it on. So that would be over to you @nepomukmaier ;-)

rotcehdnih, 2016-12-21:
Is this a issue ? im sure the code is there and you just need to use "SERIALRX_XBUS_MODE_B" . Please test and report back

nepomukmaier, 2016-12-21:
@atomiclama: Thank you! Testing is no problem. I have the equipment.

I have tried SERIALRX_XBUS_MODE_B and it is not working with the new FLEXX receivers. To my knowledge there are two different SRXL versions. One with 12 channels and one with 16. The one with 12 should work SERIALRX_XBUS_MODE_B in betaflight (I haven't tested it, but some guys on the internet reported this). The new FLEXX receivers use the 16 channel version. 

For this reason cleanflight was changed. This is the request for cleanflight (I was not involved there, as I have just started to use cleanflight & betaflight)
https://github.com/cleanflight/cleanflight/issues/681

This is what I have found in the updated cleanflight code (as an example from /rx/srxl.c):

> #define SRXL_CHANNEL_COUNT_A1 12
#define SRXL_CHANNEL_COUNT_A2 16

rotcehdnih, 2016-12-21:
Oh yes i see now , should be a simple add. i'll see what i can do :)

nepomukmaier, 2016-12-22:
Thank you! Most (all?) of the code should be already in cleanflight. :-)

nepomukmaier, 2016-12-25:
I have managed to add the 16 channel version by changing some lines according to

https://github.com/cleanflight/cleanflight/pull/1308

I hope you have not put too much effort into it.

closed 2016-12-25 by nepomukmaier

Juan-ma, 2017-04-14:
Hello,

I have a Revo board with BTFL 3.1.7 and a Multiplex RX5 M-Link configured as SRXL receiver. I have configured Port 2 (UART1) as "Serial Receiver" and in the Configuration section, Receiver as "Serial-based receiver". In the "Serial receiver provider" I have tryed both SRXL and XBUS_MODE_B with no success. With XBUS_MODE_B provider selected, I see all chanels fixed at 1500 in the receiver section. With SRXL provider is selected, when I move right stick all channels move randomly.
I have tryed with older versions of betaflight even with cleanflight but the same thing happens. I have sniffed the serial output of the receiver and it's OK accord the Multiplex description of the protocol. I have tryed with other serial ports of the Revo with the same result.
I have had a look to the code and it seems that I should select "XBUS_MODE_B" for the Multiplex receiver and SRXL seems to be used with Spektrum receivers:

https://github.com/betaflight/betaflight/blob/master/src/main/rx/xbus.c#L62
https://github.com/betaflight/betaflight/blob/master/src/main/rx/rx.c#L219

Maybe the problem is the baudrate of the serial port in the revo (multiplex receiver is 115200).

Juan-ma, 2017-04-14:
As this issue is closed, i have opened a new issue:
https://github.com/betaflight/betaflight/issues/2884