betaflight/betaflight #1852: Make CG offset parameter for best performance

nhadrian, 2016-12-18:
Hi again,

I have another suggestion. My quad's CG is out of the ideal position (given by geometry). 
This is a quite common issue because of the limited space on race frames.
And this error causes small pitching when really fast throttle changes. For example, when copter is tail-heavy, high throttle will cause nose to move up a bit. I repeat again, this issue only exists on really fast throttle changes when PID controller can't follow the high torque on picth axis. And of course this behaviour is more intensive on over-powered race copters.

Of course I was able to correct this problem by defining a custom mix on my copter. The fact that mix table calculated uppon CG geometry correction is working proves the theory above.

So my suggestion would be to define a CG correction parameter that can handle this 'CG' to 'ideal geometry center' error for any types of frame and can simply measured and calculated.
For example this could be a percentage value, from 0 to 50% (should be limited under 100% to avoid total useless configurations because 100% would deactivate some motors) and calculated based on the actual geometry:

CG_offset (in %) = (measured CG offset from center) / (front or rear motor to center distance in the same direction).  

Here is a sample custom mixtable for an X-type frame if copter is tail-heavy:
```
mmix 0 1.000 -1.000 +1.000 -1.000
mmix  1 ((100-CG_offset)/(100+CG_offset)) -1.000 -1.000 1.000
mmix 2 1.000 1.000 1.000 1.000
mmix 3 ((100-CG_offset)/(100+CG_offset)) 1.000 -1.000 -1.000
```

Here is a sample custom mixtable for an X-type frame if copter is nose-heavy:
```
mmix 0 ((100-CG_offset)/(100+CG_offset)) -1.000 +1.000 -1.000
mmix 1 1.000 -1.000 -1.000 1.000
mmix 2 ((100-CG_offset)/(100+CG_offset)) 1.000 1.000 1.000
mmix 3 1.000 1.000 -1.000 -1.000
```

**((100-CG_offset)/(100+CG_offset)) is calculated based on force and torque balance in CG.**

I don't know the Betaflight code in details but I'm sure this correction can be inserted somewhere else than individual mix tables.

Another thing is that this offset is important for the throttle amount of each motor but also could be used for pitch correction for improved pitch behaviour.

regards
Adrian

borisbstyle, 2016-12-18:
If your copter only pitches on fast throttle changes than issue is not directly the CG.
PID controller is perfectly capable to compensate to some slight CG offsets without any special mix.

We do have a solution for the fast throttle changes and pitching effect, which is still being optimised

nhadrian, 2016-12-18:
I have about 6% error in CG to geometrical center. 
I don't know if it is slight or not, but calculated mix table eliminates pitching in my case. 
Please note that I'm using 3.0.1 so perhaps pitching is not an issue in 3.1 anymore...

borisbstyle, 2016-12-18:
4% is not much at all :)

nhadrian, 2016-12-18:
actually 6% :)

borisbstyle, 2016-12-18:
Still not much and should not cause issues. It only can make "pitching" worse.

nhadrian, 2016-12-19:
Ok, thanks for your reply and feedback.

closed 2016-12-19 by nhadrian