betaflight/betaflight #1855: ESC DYS XS30A can not be calibrated using FC

Steadyspike, 2016-12-19:
This issue is coming from #1838. Suggestions coming from Borisbstyle were not valid.
I can calibrate DYS XS30A ESC using servo tester but I can not do the same using Betaflight or BLHeli suite (max throttle slider, power on, min throttle slider...). The fligh controller is SPF3EVO and the settings are:
ESC/motor protocol PWM
PPM min throttle 1148
PPM max throttle 1832
Minimum command 1000
Maximum throttle 2000
Betaflight 3.0.1
BLHeli suite 16.5.14.8.0.0

I have compared the PWM signal generated by servo tester and FC (using an osciloscope):
- Voltage: 5V (servo tester) vs 3V3 (FC PWM) -> it's not the problem because ESC detecs a "1" in case voltage is > 1.5V 
- Time slot in "High": servo tester calibrate in case the time slot >1.5ms vs 1.8 ms (FC at max throttle - screenshot in #1838) -> Initially I though it was the problem until I validate servo tester also could calibrate in the 1500-1800ms.
- PWM refresh time: 20 ms -50 hz (servo tester) vs 2.25 ms - 450 Hz (FC PWM)

So it seems the only difference is the PWM refresh time. Does it has sense DYS ESC require a value lower than 450 Hz in order to enter in programming mode? 

In case it helps, find attached the dump.
[DUMP.txt](https://github.com/betaflight/betaflight/files/662373/DUMP.txt)

nathantsoi, 2016-12-20:
i know this doesnt solve the root issue, but maybe you can upgrade to dshot? https://www.rcgroups.com/forums/showpost.php?p=36224034&postcount=232

mikeller, 2016-12-20:
Or even if DShot is not an option, I would recommend you try using OneShot 125 or MultiShot. PWM has become a 'niche' protocol for multirotor ESCs, so I would not be surprised if the limitation in refresh time was existent in the blheli_s firmware, and I suspect this would go largely unnoticed.

Steadyspike, 2016-12-20:
Dshot: In case I can not figure out where is the problem I will try Dshot.

Oneshot125/42 & multishot: I'm not sure calibration can be done using these protocols. In any case, I have tried to calibrate ESCs with these protocols without success.

mikeller, 2016-12-20:
@steadyspike: Yes, calibration can definitely be done with these protocols, even more so, a recalibration *should* be done after changing protocols, to make sure the ESC uses the right timings for the new protocol.

AndersHoglund, 2016-12-20:
This can not possibly be a BF firmware problem. You have even confirmed with the scope that you have PWM1000 signals on motor outputs, even though not quite exactly in the 1000...2000 range. But that should not matter, the ESCs should calibrate to what is coming out.
Try skipping calibration all together. Set fixed "PPM Min/Max Throttle" values 1000...2000 in BLHeliSuite and do a testflight. Maybe on the bench without props first, to setup min_command, min_throttle and max_throttle to reasonable values. 
Or, as suggested, go for Dshot.

borisbstyle, 2016-12-20:
Sorry I will have to close this one again as the signal generated from the firmware is correct like @AndersHoglund says. That is also the reason why the other issue was closed.
This totally looks like ESC hardware/software problem.

closed 2016-12-20 by borisbstyle