betaflight/betaflight #1859: Falling down on Yaw movement

tedbarth, 2016-12-20:
I'm using a BeeCore F3 with Betaflight 3.0.1 and the BRUSHED motor feature.

When I'm yawing my quad is loosing height. I guess this is because two cw or ccw motors are lowered in velocity to produce torque. Is it possible to set up the quad to speed up the others motors more to counter a drop in height? Coming from the original Tiny Whoop this is a frustrating behavior.

borisbstyle, 2016-12-21:
When you would speed up the others you would not be yawing ;)

I think you only lose height when you are yawing in turns. That is a normal behaviour due to various reasons. You have to learn compensating throttle in those scenarios.
Also this all depends of power/weight scenario of the quad. Some will even gain height when yawinf depending of how fast you yaw.

closed 2016-12-21 by borisbstyle

tedbarth, 2016-12-21:
Thanks for your fast answer!
I hit the behavior of sinking when I'm yawing - No matter if I'm standing still or moving.

> When you would speed up the others you would not be yawing 

I don't get it. To produce torque one or better two either cw or ccw (dependent on the torque direction) motors must spin faster than the rest of the motors. This should be done wile manipulating the speed of ALL motors (some go slower, while the others go faster) to not cause sinking or climbing in height while yawing. Since my quad is sinking the FC probably is reducing the speed of some motors but not increasing the speed of the rest of motors.

This might not be implemented in betaflight but it is in TriFlight (Cleanflight for Tricopters), the original Inductrix FC and at least two other quads I got in touch with (don't know what FC has bee used). I wonder why this does not apply to betaflight as well?

Or is my understanding of the physical world wrong? (I hope not)