betaflight/betaflight #1864: Add D-Shot features please (Meta-Features)

llambkin, 2016-12-21:
Hello Team,

Felix has added things like motor rotation choice, beacon, 3dmode etc.

Can we please enjoy some of this goodness in Betaflight. Maybe we can also pass some of this stuff to BLheli also, so that those ESC's can get those features also.

Thanks -Liam

Jahnkeanater, 2016-12-22:
BLHeli already supports all of these features and can be easily changed by using passthrough and the new chrome blheli configurator app. Betaflight 3.1 already has Dshot. 

You will have to wait for the next stable release or you can download and install the latest beta build. I recommend just waiting since not all boards support Dshot at the moment and a lot of features have been disabled to get Dshot working.

borisbstyle, 2016-12-22:
@Jahnkeanater This is not about passthrough. It is about activating / deactivating features without the use of passthrough.
For example you could enable/disable 3D mode on AUX switch. Or you could let the ESC's beep on AUX switch or some other scenario.

DerDominik, 2016-12-23:
As DShot works with digital signals, maybe it is possible to completely configure the ESC from the FC?
So the ESC will left with default settings and BF sends all needed settings after startup to the ESC. But I don't know if that is generally possible and how long it would take to send all configured settings?!

llambkin, 2017-02-19:
Any interest in getting the beeper happening through KISS ESCs?

mikeller, 2017-02-20:
@llambkin: #2393 adds support for the DShot beeper commands, try `escprog ds 255 1 2 3 4 5 5 5 4 3 2 1 2 3 4 5 5 5`. :-D

What use can you see for the ESC beeping, besides amusement value?

andymac4182, 2017-02-20:
Additional beeping if you damage your beeper and are looking for a downed quad or playing tunes when you win a race 😉

mikeller, 2017-02-20:
I am a bit reluctant to add it in this way, since beeping the ESCs disables motor contol, and I wouldn't want to see people's craft fall out of the sky just because they flipped the 'beeper' switch. :-)

borisbstyle, 2017-02-20:
@mikeller I think it should be possible. You can simply skip some motor commands without side effects

llambkin, 2017-02-20:
I would think there'd be the ability to only use beeper whilst the quad is
disarmed. I wouldn't see any need to have the beeper going while flying.

mikeller, 2017-02-20:
I don't know the reason for it, but currently the beeper mode can be activated irrespective of the arming state.

mikeller, 2018-02-01:
Implemented by #2393 .

closed 2018-02-01 by mikeller