betaflight/betaflight #1870: OSD SW

digifaps, 2016-12-22:
Hi, 

Can it be that we have no more option to change the osd layout of external osd's over a 'mode' slider ?

2.9 still had it. now it seems to be gone.

jflyper, 2016-12-23:
I think it's a bug introduced when BFOSD was first added.

The OSDSW mode is only available when FEATURE_OSD is enabled, but the OSDSW has nothing to do with the FEATURE_OSD.

mikeller, 2016-12-23:
@jflyper: Yes, more or less. `FEATURE_OSD` is used for all OSD related functionality, i.e. it controls the OSD switch **and** SW OSD.

@digifaps: If you enable the OSD feature on the 'Configuration' tab, the OSD switch will become available.

jflyper, 2016-12-23:
@mikeller 

I believe the OSDSW is not related to BFOSD (the integrated OSD) but is a MSP response option (BOXOSD).

I propose to register BOXOSD unconditionally as it was prior to 3.0. (Will submit a PR.)

mikeller, 2016-12-23:
Fair enough. I don't know what the intention of whoever added BOXOSD to FEATURE_OSD (@nathantsoy?) was, I always assumed there was one. But I don't care either way.

jflyper, 2016-12-23:
@mikeller Do you mean that the FEATURE_OSD has been used in the past to control BOXOSD activation in the past??? As far as I could research, it was first introduced by the initial support for sirinfpv (8a93041).

mikeller, 2016-12-23:
@jflyper I thought it had been. But turns out I am wrong, and it was the other way round: https://github.com/AndersHoglund/betaflight/commit/aacfe7b9911caabda8fed6968aba9110733c9e64#diff-b2b29e72f4b7263e160784dff0c923d7L518 . @theangularity would be the person to ask why BOXOSD was added to FEATURE_OSD in the first place.

jflyper, 2016-12-24:
@mikeller Okay, so the PR is valid...

mikeller, 2016-12-24:
@jflyper: If BOXOSD was added to FEATURE_OSD without there being a reason for it then yes.

borisbstyle, 2016-12-25:
Fixed in #1873

commit 100746cc (2016-12-25): Merge pull request #1873 from jflyper/bfdev-fix-boxosd

Activate BOXOSD unconditionally (Fixes #1870)
closed 2016-12-25 by borisbstyle