betaflight/betaflight #1878: cc3d atom doesnt arm

asaf123098, 2016-12-23:
hi, i have beta flight 2.9.1, and my atom doesnt arm (although i set it) any ideas? 
thanks!

andre92, 2016-12-23:
- check cpu load (should be well below 50% when disarmed)
- provide a cli dump

asaf123098, 2016-12-23:
hi, my cpu load is 2%, all the featurs are disabled and the gyro update frequency is 1khz.

CLI Dump:

# dump

# version
# BetaFlight/CC3D_OPBL 2.9.1 Aug  2 2016 / 22:47:13 (0ef6c3d)
# dump master

# mixer
mixer QUADX
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO_RATES
feature RX_SERIAL
feature FAILSAFE
feature ONESHOT125
feature SUPEREXPO_RATES

# beeper
beeper GYRO_CALIBRATED
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
beeper ACC_CALIBRATION
beeper ACC_CALIBRATION_FAIL
beeper READY_BEEP
beeper MULTI_BEEPS
beeper DISARM_REPEAT
beeper ARMED
beeper SYSTEM_INIT
beeper ON_USB

# map
map AETR1234

# serial
serial 20 1 115200 57600 0 115200
[... 267 lines cut ...]
set i_level = 50
set d_level = 100
set p_vel = 55
set i_vel = 55
set d_vel = 75
rateprofile 0

# rateprofile
rateprofile 0

set rc_rate = 100
set rc_rate_yaw = 100
set rc_expo = 10
set rc_yaw_expo = 10
set thr_mid = 50
set thr_expo = 0
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 70
set tpa_rate = 20
set tpa_breakpoint = 1650

# 

thanks!!

jflyper, 2016-12-24:
Throttle low end point is 1100 or greater.

AndersHoglund, 2016-12-24:
Tx throttle that is. 
All sticks low endpoints MUST be below "min_check" and high end point above "max_check", not only the throttle stick. Usually you should adjust Tx end points to 1000 and 2000. On some Tx this is not possible, use cli command "rxrange" in that case.

asaf123098, 2016-12-24:
my tx's min points are 1116+- and my min check is set to 1100, tried to check rxrange and it is set to 1000 -2000, how do i change my tx range?

AndersHoglund, 2016-12-24:
Nonono. Not above 1100. Below. 
As said before, low stick endpoints should preferably be 1000. If you can not change Tx endpoints, use cli command "rxrange 0 1116 2345" Where "2345" is you stick max endpoint. Repeat for all channels 0,1,2 and 3. Then "save".
Or, "set min_check = 1200" . Try one of those, one at a time. Not both at the same time.

borisbstyle, 2016-12-25:
Closing issues for 2 reasons:
1) old firmware version
2) sounds like incorrect configuration. Please configure your board and tx properly

closed 2016-12-25 by borisbstyle

asaf123098, 2017-01-02:
AndersHoglund

> Nonono. Not above 1100. Below.
As said before, low stick endpoints should preferably be 1000. If you can not change Tx endpoints, use cli command "rxrange 0 1116 2345" Where "2345" is you stick max endpoint. Repeat for all channels 0,1,2 and 3. Then "save".
Or, "set min_check = 1200" . Try one of those, one at a time. Not both at the same time.

tried both options, when arming, all the motors start working (in the tx the throttle is idle), and if i try to use my throttle or yaw, pitch or roll, it responds, but i cant get the motors to stop working (idle).