betaflight/betaflight #1887: [Feature Request] Low voltage warning delay?

braeden123, 2016-12-24:
OpexTX's telemetry has an option to only activate on sustained voltage drop, unlike betaflight which will activate on voltage sags and punch outs.

Credit: https://www.reddit.com/r/Multicopter/comments/5k2gkj/is_there_a_way_to_set_a_delay_for_low_voltage/

mikeller, 2016-12-26:
Sounds like a reasonable suggestion.

digitalentity, 2016-12-26:
I agree, this will be a very nice thing to have.

Eganwp, 2017-01-06:
Just don't mix-up "delay" with "duration". It's duration of low voltage over X timeframe we want before the logic switch is triggered, not delay. I often see this mistake made on videos and online guides people make for the Taranis on OpenTX VFAS logic switch. ;) Just a heads-up.

laz2wiringpi, 2018-11-01:
while i do not agree with the implementation duration for voltage warnings #6691
the warnings for voltage drive me nuts it was not as bad in 3.3 
 
i plan to use  both a delay and  baterryvoltagelatest to detect a  recovery of the battery  

currently i use a hacked  baterryvoltagelatest  in the telemetry and do it on my transmitter.  for the delay of a low battery an this has worked out for me .

mikeller, 2019-04-28:
@braeden123: This should have been implemented by #8012. Can you verify / close please?

closed 2019-04-28 by braeden123

Combinacijus, 2023-01-03:
In CLI paste
```
set vbat_duration_for_warning = 20
set vbat_duration_for_critical = 5
save
```

This means the voltage has to be below the warning voltage for 2 seconds before the warning is shown, and half a second below the critical voltage before the "land now" message is shown.
You probably don't want to go above 3 sec to not kill your battery