betaflight/betaflight #1889: #842 GPS orders in CLI doesn't work - SPRACINGF3

nhadrian, 2016-12-25:
Hi,

today I uploaded #842 to my Beerotor FC (SPRACINGF3) to test D-Shot.
It looks like something is broken inside CLI because none of GPS orders are working (invalid command as reply). Of course serial port is configured and GPS is enabled in configurations panel.

I'd like to set up `gps_auto_config = ON` for my NMEA GPS.
Without this setting I can't test if GPS parser is working or not.

Regards
Adrian

borisbstyle, 2016-12-25:
GPS has been removed on F3 targets in order to speed up the code.

closed 2016-12-25 by borisbstyle

nhadrian, 2016-12-27:
OK, but then why GPS function itself is still being active?
Sending `functions` still shows that GPS is active if enabled by GUI.
Quite deceptive.

So clearly spoken, I should by at least F4 controller for having GPS coordinates in telemetry or in OSD displayed after official 3.1 release?  
GPS coordinates are really useful for lost model finding!!!

Regards
Adrian

mikeller, 2016-12-28:
I assume you mean `features`? The fact that all features are present in all targets irrespective of there being hardware / software to support a particular feature is a shortcoming of the current implementation of configuration management. It is also something that is scheduled to be improved with the introduction of parameter groups in 3.2.