betaflight/betaflight #1891: Angle mode doesnt work...

sm5hua, 2016-12-25:
Hello all

I just found out using beta #843 that angle mode doesnt work properly. Neither roll nor pitch works. I am using a Kiss FC and dshot600. Acro and horizon works fine but as soon as I switch to angle, pitch and roll stops working, When I look at my taranis and in betaflight as well, it shows it works but not in real life. 

Just wanted to tell..

Best wishes: Peter Laur, Sweden

jflyper, 2016-12-26:
@sm5hua #843? is way back in July. Did you mean #1843?

sm5hua, 2016-12-26:
Yes I did.. And corrected it now.. :)

sm5hua, 2016-12-26:
Just installed #1846 and angle mode pitch and roll still doesnt work. Hope someone else has the same issue. When I look in the betaflight configurator its shows as if they work but no response at all. And is only angle mode, Acro and horizon works..

martinbudden, 2016-12-26:
Do you really mean https://github.com/betaflight/betaflight/pull/1846 ? That was a PR that wasn't merged in.

Have you tried the latest build at: http://andwho.sytes.net:8080/job/BorisB_BetaFlight/lastStableBuild/

sm5hua, 2016-12-26:
I mean the lastest beta of course. Will change it..

borisbstyle, 2016-12-30:
@sm5hua You seem the only one reporting this issue?

sm5hua, 2016-12-30:
Hi Boris. Yes seems so :) I have tried several betas after the one I reported and still it doesnt work. Now I will try backwards instead. I know all flight modes worked for me about a week before this beta. Will get back and report when it worked and when it started not to.

The odd thing is that in betaflight, I see pitch and roll works but no reaction in the quad when in angle mode. In acro and horizon it all works. I am using a Kiss FC here and the 24A race ESC and dshot600. Have tried oneshot125 and no difference...

borisbstyle, 2016-12-30:
can you reset your configuration and reconfigure

sm5hua, 2016-12-30:
I will do.. Its gonna take a while... Might be some problem in my saved config so doing it all from scratch is probably the best. Will get back to you soon..

sm5hua, 2016-12-30:
Haha! It works now! It was a configuration error. I did load my saved config and now when doing it all from scratch, it works :)

borisbstyle, 2016-12-30:
Great !

closed 2016-12-30 by borisbstyle