betaflight/betaflight #1898: Battery monitoring throttle reduction

Yamaford, 2016-12-27:
With brushed microquads becoming more popular and no osd or telemetry, would it be possible to use the battery monitoring function to say cut the throttle in half(3/4?) at a certain voltage? This would give a physical indicator of a low battery and maybe prevent damaged lipo or un controllable decent. If the throttle is cut but you still have control you can land.

mikeller, 2016-12-28:
@Yamaford:  `vbat_warning_cell_voltage` and `vbat_min_cell_voltage` were designed exactly for this, and allow you to use a beeper / LED strip to indicate low / critically low battery voltage.

Yamaford, 2016-12-28:
I know on bigger models it works great. Im talking about these sub 70mm
ones that will be showing up soon. No room for buzzers or LEDs and wiring
to go along with them. Every gram counts. LEDs dont do you any good FPV.

mikeller, 2016-12-28:
@Yamaford: I find that addition of a micro buzzer (soldered straight to the pads) is a useful addition even to micro quads, and they come in at just 0.5 grams or so. But ok, if you don't want to do that then throttle reduction might be a way to indicate battery voltage drop. Even though I doubt that you'd notice the effect of this when flying FPV before hitting the ground, it could still be valuable to reduce the potential for battery damage.

Some of the concerns I have with this suggestion:
- it introduces the potential for oscillations: Battery voltage drops -> throttle is reduced -> battery voltage raises due to current -> throttle is increased again -> rinse, repeat. I think to make this work, the throttle reduction would have to be made permanent until the craft is disarmed (at least);
- it works in a counter - intuitive way: If you are flying at a battery level that approaches the 'empty' voltage level, and are forced to do a hard maneuvre to avoid a collision, the increased current consumption and associated voltage drop will cause throttle reduction, potentially leading to exactly the crash that you'd have been able to avoid without the throttle reduction.

Also, I personally find that for comparable flight profiles, battery capacity is fairly well correlated to run time, so setting up (conservative) timer limits on your TX should go a long way towards avoiding battery damage.

That said, it would be easy to implement the throttle reduction.

DanNixon, 2016-12-28:
I do worry that voltage sensing of a 1S cell is not particularly stable/accurate enough to do this.
I have a telemetry enabled LT105 and the voltage varies far too much to use it as a measurement of remaining battery capacity.

martinbudden, 2016-12-28:
I find with micro brushed quads you can feel the battery is getting low by how it reacts to stick commands.

ledvinap, 2016-12-28:
@martinbudden : Maybe exacerbating this effect will help - decrease motor command on lower voltages. Lowpass filter should help with feedback loop/oscillation.

mikeller, 2016-12-28:
@ledvinap: My choice of words might have been poor when I said 'oscillations', what I suspect might happen would be oscillations in the order of 1 Hz or so - so a lowpass would probably not work to get rid of it.

ledvinap, 2016-12-29:
@mikeller: Yes, I understood the problem. Few seconds of time constant should be fine with respect to battery discharge. And user must compensate by increasing throttle command, so motor command will increase proportionally only to battery discharge.

borisbstyle, 2016-12-30:
I was planning to add something like auto-land feature on low battery. The reason is because some of the stock firmware on these small brushed quads already has that. If the user is used to it than it is a nice feature to have

MJ666, 2017-01-03:
With the small 1S powered micro copters we operating batteries at the limits and usually have very high voltage drops. We did quite some testing in the past and found even voltage monitoring not very useful for tis class of copters. Some of our brushed AlienFlight controllers have voltage monitoring and even buzzer ports. For many of my brushed builds i can punch out with an fresh battery and already hit the low voltage alarm. I likely would not use this feature and stay with an timer in my TX.

SebiTimeWaster, 2017-03-21:
i just flew a battery down to 2.1 volts because i expected this feature to already exist. (also i didn't find an exact description what is supposed to happen when the limit is reached, my copter has no buzzers, lights, osd or telemetry)

the problem with the voltage depending heavily on the current load is normal and can actually be easily fixed with a rolling buffer that normalizes the voltages from the last 10 or even 20 seconds, all drops produced by heavy load should be filtered out that way and 20 seconds accuracy for a low voltage shutdown is enough to not hurt the battery.

the problem of "suddenly dropping out of the sky" can be avoided by applying an multiplier on the throttle that falls from 1 to 0 in lets say 3 seconds, that way it should land softly enough, the mini-copters can actually take quite a beating.

i really hope these features get added, atm i don't know how to avoid this problem.

Engineereric, 2017-04-23:
I found this thread because I was looking for exactly the opposite. I find the drop in throttle response even half way through the battery profile to be enough to stuff up my muscle memory for flying the real thing.
I would like to see PWM compensation to keep the throttle response the same through the entire battery discharge curve.
If some one could point me to where in the code I should put that would be much appreciated as I am happy to implement it.
regards
Eric

mikeller, 2017-04-24:
@Engineereric: What you are asking for is not possible, unfortunately. The maximum rate of acceleration for the motors is dictated by the voltage that the batteries can supply at any given load. There is no way to 'compensate' for the voltage drop on a not fully charged battery.

nathantsoi, 2017-04-24:
are you looking for `vbatPidCompensation` (cli: `vbat_pid_gain`)
@Engineereric?

Engineereric, 2017-04-24:
No.
I am looking for VBat PWM compensation, back off f PWM with higher VBat so that the power curve through the discharge cycel is flatter. 
Its actually VBat^2 function but basically want to compensate to effectively regulate the Voltage to the motors equivalent to say 3.3V then you can set up your rate and it remains constant.
Same as a Halogen light controller.

nathantsoi, 2017-04-24:
Nice, sounds like a really cool feature. PRs are always welcome! Lmk if
there is anything I can do to help get you started, but maybe grepping the
code for the vbatpid stuff is a good starting point?

mikeller, 2017-04-24:
Ha, good point @Engineereric, if you are willing to sacrifice your maximum acceleration with a full battery then it is very well possible to achieve the effect that maximum acceleration stays constant over the whole battery range.

Engineereric, 2017-04-25:
Yeah so I thought I'd have a go at building this project. 
I normally work with PIC and ATMEL devices as my day job, so didn't think it would be very hard.
Man was I wrong!
The guide on I found at 
https://github.com/martinbudden/betaflight/blob/master/docs/development/Building%20in%20Eclipse.md
is incomplete and very wrong in many places.
Can some one just tell me which IDE to install and the project folder to copy so I just need a few clicks to get up and running or is this one of those things cobbled together with free stuff that I need to hold my tongue in just the right way to get it to build?

AndersHoglund, 2017-04-25:
Hard?  

    git clone https://github.com/betaflight/betaflight-configurator.git
    cd betaflight
    make help
    make make arm_sdk_install
    make SPRACINGF3EVO 

As for IDE, your personal pref will do fine as you are into this business already and you would know what to do. bash, grep and emacs will take you far.
Eclipse with CDT, gdb and  SEGGER J-Link debugger plugins, etc etc will take you further. However, I find Eclipse a config-nightmare, especially on this type of project with 70+ configurable targets. Some like VisualGDB. J-edit is a strong candidate too.

lostcontrol, 2017-04-25:
Of course, you want to:
`git clone https://github.com/betaflight/betaflight.git`
and not the configurator web app.

AndersHoglund, 2017-04-25:
Yes, of course. Copy&paste error, corrected.

Engineereric, 2017-04-25:
Yeah I know I was afraid of that.
Linux is easy in what you describe there but I have not found a real integrated editing environment.
As for the IDE in Windows you are right about Eclipse it does my head in.

I’ve used AC6 which is based on eclipse but works out of the box so will give that a try I guess.
Just would have been nice if something like the MPLAB or AVRstudio existed where you just click on the project file and if you copied the right folder it just all works.
I don’t think I’d have a job if I told my clients just build it in which ever environment you are happy with and use the configuration that best works for you!

None the less the developers of Betaflight have done an amazing job and the fact that so many people can use it with out any techincal background is a real credit to the developers.

Thanks for your help.

Regards
Eric

From: Anders Höglund 
Sent: Tuesday, April 25, 2017 9:44 PM
To: betaflight/betaflight 
Cc: Engineereric ; Mention 
Subject: Re: [betaflight/betaflight] Battery monitoring throttle reduction (#1898)

Hard?

git clone https://github.com/betaflight/betaflight-configurator.git
cd betaflight
make help
make make arm_sdk_install
make SPRACINGF3EVO 
As for IDE, your personal pref will do fine as you are into this business already and you would know what to do. bash, grep and emacs will take you far.
Eclipse with CDT, gdb and SEGGER J-Link debugger plugins, etc etc will take you further. However, I find Eclipse a config-nightmare, especially on this type of project with 70+ configurable targets. Some like VisualGDB. J-edit is a strong candidate too.

DanNixon, 2017-04-25:
> I don’t think I’d have a job if I told my clients just build it in which ever environment you are happy with and use the configuration that best works for you!

I would love to be told that whenever I start working on a new code base.
IMO things like IDE/editor and debugging tools should be the developers choice.

AndersHoglund, 2017-04-25:
It all depends what you want to do. 

If you only want to "build the project" there is no need for an IDE. You need some kind of Linux env, Real, virtual or emulated. Like Vagrant (there is a ready made Vagrantfile to use),  Ubuntu in a Virtualbox, Cygwin etc etc. There was also some work done on setting up a docker env, not sure about the state of this. Some info here: https://github.com/betaflight/betaflight/issues/1538

If you want to develop and contribute, there is no limit to what env you run your IDE, editor or debugger in. Freedom of choice.

mikeller, 2017-04-27:
@Engineereric: Have a look at https://github.com/betaflight/docker-betaflight-build . All that this requires as the local installation is docker in order to get your build environment up and running.

Engineereric, 2017-04-27:
So loosing hair and sleep over Eclipse. That is just not working for me. Seems I cant even get to building the code because it has to many errors with its plug-ins.
As for Docker, not supported on my machine and from what I see its only another environment like LINUX , not an IDE which is what I want. 
Only real way to efficiently work on new code is in an IDE where every thing is visible and editable at the same time.
Thanks for every ones help, I will have to revisit this at a later stage because I have already spent to much time on it and cant even get the original clone to build.
This page:-
https://github.com/martinbudden/betaflight/blob/master/docs/developmen/Building%20in%20Eclipse.md
Should probably be deleted as it is so wrong it is just stupid. Probably years out of date.
Or just way too much missing.
If I got it to work I'd have happily written one but no joy.
Thanks Every one.
The whole plan was to be able to run 2cell safely on 3v motors and keep the performance stable.

Engineereric, 2017-04-27:
So I got every thing installed and linked I think in Eclipse, finally all the environment variables are set.
Now it builds sort of with 43 errors and 4 warnings!
I thought the clone is supposed to be a working copy?

AndersHoglund, 2017-04-27:
C'mon. You do not build in Eclipse. You setup Eclipse to run "make ", on some local or remote Linux-like host env. Or just run make in a shell....
As I tried to say, building is one thing while editing/designing/debugging is another. This is a "MAKFILE" based project, not tied to and limited to any specific/propriatory IDE env.  Freedom.

AndersHoglund, 2017-04-27:
To have a working build env in Eclipse, you would need to duplicate everything done by the makefile exactly. But as you are a "programmer", you know that duplication is a bad/stupid idea. 
I.e. Do not duplicate it, use it. Just as the "copy&paste" programming style is bad. Maintaining copies and duplicates is a PIA.
And if docker does not work for you, try vagrant.

Engineereric, 2017-04-27:
Look I dont need to be wound up any further this has been frustrating enough thank you.
I'll stop bothering every body and go and do something else sorry for offering my time.
Sorry.

mikeller, 2017-04-27:
@Engineereric: What target were you trying to build? Since there are over 100 targets that Betaflight builds now, the Makefile requires a TARGET=<target name> parameter in order to build the correct target.

AndersHoglund, 2017-04-27:
@mikeller Not "requires", rather "accepts". This is the old legacy way, still supported, but now you can just name your target as an arg, or even multiple targets in a list. E.g.  `make REVO SPRACINGF3 CC3D NAZE`
or `make REVO_clean REVO ` for clean rebuild.

mikeller, 2017-04-27:
@andershoglund: Your examples are for running the build with `make`, which is different from building with Eclipse, which is what @engineereric is trying to do. Run this way, not setting a target will fail, and setting the TARGET variable in a build step will lead to defines not being properly evaluated on the IDE.

AndersHoglund, 2017-04-28:
@mikeller My comment and examples was of course on `make`, based on your statement on `make`: 

> the Makefile requires a TARGET= parameter

What is needed to build in Eclipse is another matter. A few different options exists, including running make with the target specified in either of those two ways. I only reacted on the word "requires", not quite true.

AndrewFarley, 2017-10-23:
With the 1s quads becoming ever-so-popular now, I landed here also looking for what @Engineereric was looking for, increasing the throttle response (at the cost of max throttle) when the voltage is lowered.  Did someone file a separate bug to get visibility into this desired feature?  It's something I might work on if I have a few spare cycles soon.

nmw01223, 2018-02-12:
I'm after what AndrewFarley and others have said, for the same reason.

I've never looked at the code, but I'm guessing thr-req comes in and (alt hold aside) is passed pretty much as is to the algorithms with gyro/accel etc that then generate the motor demands? I guess it would pretty much be a square law, so one could modify thr-req% to thr-dem% = ((NomV / ActV) ^ 2) * thr-req%. Would need to clip it of course, and possibly ActV would need an offset to allow for driver device saturation voltage.

The closer it is to the raw throttle input, the nearer it will be to what the user actually does. It cannot provide more than vbat actually allows.

Just a thought.

bnmoore, 2018-03-25:
@Engineereric @AndrewFarley @nmw01223 
I landed here in search of the same.  Got a build environment working (Visual Studio Code) and submitted a pull request.
https://github.com/betaflight/betaflight/pull/5528

nmw01223, 2018-03-25:
Interesting. I have just set up an environment under Linux (Ubuntu 14.04 
LTS). Cloned the repository and had a quick look at the code. Built some 
targets (Omnibus, OmnibusF4, SPRacing) OK but not yet tested anything 
while I wait for a new FC (Omnibus F3) to arrive to practice on (don't 
want to trash an existing one).

Completely unfamiliar with the code, but looks to me that adjustments 
could be placed in and around fc_rc.c#310-325, updating 
rcCommand[THROTTLE] by picking up the average cell voltage from 
getBatteryAverageCellVoltage() at battery.c#473 (ThrottleToUse = 
ThrottleInput * (NominalCellVoltage / CellVoltage) with a few range 
checks and limits around it).

Need to add a few NV parameters though - nominal cell voltage, max 
voltage adjustment  limit, min voltage adjustment limit maybe. Haven't 
figured out yet how to add new parameters to the NV CLI list, though 
looking at contents of settings.c and pg.c.

If anyone could point me to any documentation about the overall 
structure of the betaflight software, it would save a lot of time ... ?

mikeller, 2018-03-25:
@nmw01223: There is not much in the way of architecture documentation for Betaflight - unfortunately. Also, the feature proposed in this issue has already been implemented in #3510, and this issue should be closed.

closed 2018-03-25 by mikeller

nmw01223, 2018-05-05:
With regard to what Engineereric and Andrewfarley are after, I have built some code that does exactly that. Probably needs some refinement but basically does what it is intended to. With battery between a low and high limit, broadly scales throttle (after any RC curve) as  new = old * (NomV / ActualV). Beyond limits, scales as if was at the limit. Parameters set in CLI.

Work on Furibee with F3 board (SPRACINGF3), and also an 80mm 1103 brushless F3 (OMNIBUS) with both 2s and 3s batteries. No real intention to feed these files back into github (don't know the process for a start), but happy to put the zipped sources on to Google drive for anyone to get at.