betaflight/betaflight #1899: Blackbox readout broken on TINYFISH target

fishpepper, 2016-12-27:
I just found out downloading blackbox logs from TINYFISH does not work.
Blackbox erase and writing works.

When i click SAVE file to disk nothing happens.
SPI traces show 0x03 0x00 0x00 0x00 as command+address (= bf tries to read address 0x000000).
Then i see lots of data beeing read out, SCK, MOSI, MOSI, CS all look valid. 
Approximately 1 second later the same command is repeated again and again with the same start address....
I need to investigate what happens... Any hints/ideas are welcome :)

flash_info shows (i use an M25P16 on this board):
`
# flash_info
Flash sectors=32, sectorSize=65536, pagesPerSector=256, pageSize=256, totalSize=2097152, usedSize=149504
`

DieHertz, 2016-12-27:
Yeah, I noticed it on the latest master on my SPRF3 yesterday as well. CRC error and corrupt file at output.

fishpepper, 2016-12-27:
I think my problem is different. I do not get any output at all... The download hangs indefinitely. It is trying to read from address 0x000000 all the time in an endless loop :(

fishpepper, 2016-12-27:
I tested my spracingf3 with master build. downloading works fine, file is valid.
So my tinyfish problem is something else....

fishpepper, 2016-12-27:
ok, i found out that using a usb2serial converter on uart3 works. VCP does hang.
probably related to: https://github.com/betaflight/betaflight/pull/1642

fishpepper, 2016-12-27:
seems to be fixed pull request https://github.com/betaflight/betaflight/pull/1904

martinbudden, 2016-12-28:
Closing as fixed by PR https://github.com/betaflight/betaflight/pull/1904

closed 2016-12-28 by martinbudden