betaflight/betaflight #1905: Separating IO timing from flight control timing

mkschreder, 2016-12-27:
I think it would be a good idea to separate code that interacts with io from main flight control code and run these as two separate threads. The idea being to make maximum use of the time when code waits for hardware to finish an io operation. For example then it becomes possible to run pid controller while code that reads the sensors waits for the data transfer to finish. This would allow scheduling io better and in the future it can even be partitioned further so that i2c transfers can be done in parallel with other kinds of transfers like spi and uart.. 

As for communication between the threads, queues will be used where the io thread will push latest values (overwriting whatever is in the queue) and where the fc thread can then easily check for new data being available without worrying about synchronization (it will be done in a nonblocking fashion). If data required for some part of the loop is not available then the main thread can do something else. 

This way I think the flight control can have an even better performance making most use of all available cpu cycles.

ledvinap, 2016-12-27:
@mkschreder :  This discussion was here few times (quick search for my posts: https://github.com/betaflight/betaflight/issues/1681, partially https://github.com/iNavFlight/inav/pull/619; there are others)

closed 2016-12-30 by borisbstyle

martinbudden, 2016-12-30:
@mkschreder 

>you define a gyro update as full update including dcm update, a pid iteration, mixer and output to motor or do you just mean dcm update with latest gyro data?

I define it as the full update. Last time I measured it, timings in microseconds, on a 76MHz F3 processor were:

gyro read: 12 us
gyro process and filter: 7us
PID controller: 17.5 us
RC smoothing and calculation of setpoint: 11.5 us (only run when there is new RC input)
mixer and motor update: 26.5 us (this might be slightly reduced now, since this code has been optimised a bit - maybe 2 or 3 us less)

That's about 75us in total, so we can easily run the PID loop at 8kHz (ie 125us interval).

F4 processors are considerably faster (for everything except the gyro read).

ledvinap, 2016-12-30:
@martinbudden : That assumes that gyro task gets scheduled in reasonable time. Without preemption, it will impose quite strict limit to max time slice available to tasks..

martinbudden, 2016-12-30:
@ledvinap , that's certainly true. Generally when people run at 8kHz they as many other tasks as possible to achieve this. That's fine for betaflight, but not really an option for iNav. But then people don't tend to run iNav at 8kHz.

My view is that for absolute max performance in betaflight you are better off without an RTOS. Although, as I have said before, I think well designed cooperative scheduling is much underrated, there is a stronger case to use preemption for iNav.

ledvinap, 2016-12-30:
@martinbudden : 
(I2c read of gyro at 400kHz takes ~200us without counting start/stops)

With preemption 8k will be easily possible, with almost half of CPU time left for all other tasks ...

(Sorry, repeating it again:)

I don't propose replacing cooperative scheduler; just separating 'realtime' tasks into separate thread ('full update'), so they can preempt lower priority. And possibly implementing 'idle' priority in future.

Big advantage of RTOS is bus arbitration - it will be quite easy, much simpler than pure interrupt based approach.

Another problem with current code is that some interrupt handlers take too much time (bad interrupt latency). Moving the processing to separate priority level will help a lot.

borisbstyle, 2016-12-30:
@ledvinap Can we continue this discussion here https://github.com/betaflight/betaflight/issues/1681