betaflight/betaflight #1913: Cleanup VTX RTC6705 support.

hydra, 2016-12-28:
There are two implementations, soft rtc6705 (bitbanged) and another version that uses SPI.  They differ wildly.  Suggest using the best of both of them and use the API, specialist SPI IO initialization, and the VTX POWER (ON/OFF, not RF power) support from the SPI version.

The SPI version is also used in cleanflight, PR #1912 goes some way to provide parity between BF and CF for the SPI driver.

IIRC The OSD code also needs updating a little two.

Affected targets:
SPRACINGF3NEO, SINGULARITY, SIRINFPV, FishDroneF4

Linjieqiang, 2016-12-29:
Sound great,I will pay attension to this issue.

phobos-, 2016-12-30:
A common API for both approaches sounds like a great idea, however I think leaving just one approach and getting rid of the other might be even better. What do you think?

Is the SPI version fully capable of i.e. reading registers from the chip? I think @jflyper had problems with RTC6705 not respecting CS line pulled high, so maybe hardware SPI is not the best way to interface with it.

jflyper, 2016-12-30:
I think we can deal with different underlaying transport (bus) under an unified API. Underlying transport may be hardware SPI, software SPI that can also read the SPI-ish RTC6705 registers, or even an I2C-RTC6705 or UART-RTC6705 bridge, and we can consolidate these under the unified API. 

However, I'm thinking of bit ahead; consolidating non-RTC6705 vtx including TBS SmartAudio, IRC Tramp, and PWM-controlled Aomway ones. We need a proper abstraction for this to happen. I'm currently trying to organize my experience with writing a couple of vtx drivers to imagine how we can do this.

I think we can start with consolidating the current RTC6705 drivers, but the unified API should be prepared with not only cleanup but certain extensibility in mind.

fncisco, 2017-03-18:
I have resolved the power output control to rtc6705 for the fishdronef4, but is for 3.1.6 version and 3.2 are in some heavy cleanup, must I wait to adapt it to 3.2? or someone can explain to me how commit to 3.1.6 version.

hydra, 2017-04-14:
I did a lot of work on this today:

https://github.com/cleanflight/cleanflight/pull/2721

hydra, 2018-05-12:
@jflyper this came up in conversation last week as there are 3 soft spi implementations in the codebase now.

reopening.

JBurk10, 2019-02-10:
WHY CANYT I GET MY TRANSMITTER TO TALK TO MY F3 AND SHOW THE RECIEVER ON THE BETAFLIGHT SOFTWARE...? PLESE CAN SOMEONE HELP. I HAVE 3 EACHINES 220 SITTING IN FRONT OF ME AND HAVENT BEEN ABLE TO FLY ANY OF THEM

mikeller, 2019-02-10:
@JBurk10:

- got to the nearest electronics store;
- buy keyboard;
- replace the keyboard with the stuck shift key;
- get answers to your questions.

No, seriously, it is considered offensive to send messages in all caps, and your chances of getting a response are low.

hydra, 2019-02-10:
@Jburk10 Also dont buy clones of my flight controller and you are more likely to get help.

jflyper, 2019-02-11:
@JBurk10 This place is basically NOT a place for end user support. There are couples of dozen threads in RCG for X220 which are more appropriate and where you are likely to get more help.
The most active one is
https://www.rcgroups.com/forums/showthread.php?2720753-Eachine-Wizard-X220-%28Under-construction%29

hydra, 2019-03-10:
Removed un-related issue high-jacking comments.

mikeller, 2019-03-10:
@hydra: Stop vandalising the Betaflight GitHub, and immediately reinstate my comment!

hydra, 2023-04-11:
This still needs doing, but after 7 years it seems no-one wants to do it, so closing.

lsq864, 2023-08-29:
你好，想问下TX5813怎么 address 7我传输的数据是0x4C7D，但是功率计上只有3dBm，大概是什么原因导致的呢

closed 2026-09-26 by blckmn