betaflight/betaflight #1921: Laptimer inside BetaFlight/OSD/Taranis

DaveLister007, 2016-12-29:
I have a idea of different laptimer than what is currently available. All current solutions require quite a bit of external, often pricey hardware.

How about we put IR receiver onto our copters, and place IR transmitter into gate/start/finish. That way, you will have information about your lap time inside Betaflight, and you can display it in your OSD, or send it via telemetry into Taranis which will tell you your time after you pass the gate.

I did a little research, and found these little receivers https://www.sparkfun.com/products/10266. They have integrated demodulator and filters, so it's output can be read directly in MCU. It probably wouldn't require any external hardware on your copter, just hook the receiver to +5V, GND, and some free digital input on your board (PWM, RSSI channel, etc.).

In its simplest form, we would only need to place IR transmitter into finish gate, and have it blast logical one at all times. When copter flies above it, Betaflight would detect that there is logical 1 on receiver, and track it somehow. At first, we could only send this information via telemetry to Taranis, and set up trigger there to tell you your time and reset timer. After that, we could do proper time tracking inside Betaflight and display it inside OSD. How cool would it be to see your lap time in real time while you flying?

What do you think, is this feasible? Do you see any problems?

FieserKiller, 2016-12-29:
Hi. I did that a little differently. had an arduino and a ir receiver on my quad and sent lap events to taranis over telemetry. worked great. but had no time to finish and publish, so its rotting for 2 years now. However, your approach is even better. 
to make lap detection work reliably you need a pulsed ir signal, eg with 38khz. there are receivers which recognise and decode this signal automatically, eg: TSOP1738 
No further modulated data is needed in the simplest case. if TSOP1738 closes circuit, we have flown over the transmitter. subtract last time stamp from the current and here is your lap time :)

DaveLister007, 2016-12-29:
Hey, good to hear that. I know about that line of IR receivers. I posted link to TSOP38x, there is TSOP48x, TSOP17x and probably others. Im guessing they differ in size and/or filtering that is used. Also, do you have any idea what IR receiver does when receiving constant 38kHz pulses? In datasheet, it says it filters out such signals, but I have no idea how. Investigation will be needed.

Did you run into any problems with interference from natural IR sources (eg. sun), or does the IR receiver filter it out good enough? If I rely only on blasting logical one from IR transmitter, and I get interference from sun, it would trigger false lap.

I also had and idea to use existing Current or RSSI input betaflight, and just set up trigger on Taranis to those values, since they are already transmitted via telemetry. But those pins are on ADC with unknown refresh rate, so it will probably miss the pulse. Interupt pulse detection might be needed.

FieserKiller, 2016-12-29:
Receiver goes to low on signal line when he sees 38khz and back to high after a few pulses (5-10 or so) because he thinks its interference. to transmit some real data you need to pack it into blocks on transmitter side. BUT: for simple lap counting thats not needed. copter will fly out of IR transmitter range, come back in on next lap and receiver will trigger again.
In my stupid tests (pointing with remote controls and bright bulbs on the receiver) I had pretty reliable results. however, bright sun could be a problem if it simply outshines small transmitter LEDs. people will have to get more powerful transmitters or put them unter a shady tree.  Indoors reflections could be a problem. but thats the same problems you have with all lap timing solutions.

borisbstyle, 2016-12-29:
This is a good idea, but I still think the external measurement is the way to go. The reason for that is that you can use it on any firmware and setup. Also it is easier to gather results. 

It would be nice to somehow integrate a timing system anyway to betaflight as you could receive your results and display on osd.

martinbudden, 2016-12-29:
@borisbstyle , I presumed that this was more for people practicing on their own (or with friends) rather than for actual organised races.

borisbstyle, 2016-12-29:
There are already vtx timers which work pretty much out of the box for when you race alone. 
I think this is quite some work to setup only for yourself

DaveLister007, 2016-12-29:
Of course, external solution is better for races, if you want to compare results etc. 
This would be more suited for training purposes. If you do a lap, you will immediately know your time and know if what you did was good or not. For this, this timing system is so much simpler, you just throw IR transmitter on the ground and go flying. Also this requires <5$ hardware, instead of 60$ vtx timing soluotion.

borisbstyle, 2016-12-29:
You could also simply add a switch to trigger timer / stopwatch in OSD?
If you setup a finger on a right switch you can time yourself. i think the
accuracy is not of that great importance for practice...right?

DaveLister007, 2016-12-29:
I guess you could, if you really wanted to. You could also get somebody to stand there with a stopwatch. But this way it would be so much cooler :)

I'm probably gonna try to implement it. At first only in its simpliest form where I do simple gate triggering in betaflight and leave the actual timing etc to taranis or osd, as there is already some support for this.

mikeller, 2016-12-29:
I like the idea, if it leads to a low cost solution for diy lap timing. You could even use discarded IR remotes with a button jammed as 'base transmitters', most of them are modulated at 38kHz.

> At first only in its simplest form where I do simple gate triggering in betaflight and leave the actual timing etc to taranis or osd.

Not sure if this will work, since you'd have to take delays introduced by the telemetry channel (most systems use slotted transmission). But you could simply record the current time in ms for the last timer event, and send back the delta whenever a new event is recorded (probably in the air speed field).

jflyper, 2016-12-30:
@DaveLister007 I hacked it up for you.

slawek89, 2016-12-30:
It is bad idea. I saw this kind of time measuring. If receiver was on copter and on start gate was ir transmitter. Everything was fine till you won't fly from shadow to sun light. Sun caused to much interference.

mikeller, 2016-12-30:
@slawek89: What kind of modulation did the solution that you saw use? I think the commercially available IR transmitters / receivers are pretty good in that respect, since manufacturers have realised that customers are not entirely happy if they can't use their TV sets in broad daylight.

DaveLister007, 2016-12-30:
@jflyper Thanks a lot! I just picked up all the needed hardware, so I will do some testing and report back.

About @FieserKiller concers, it really depends on what IR modulation is used. Modulation used by IR receiver mentioned here sould be resistant to sunlight, but there is only one way to find out.

jflyper, 2016-12-30:
@DaveLister007 No prob. Hope the experiment goes well.

DaveLister007, 2017-01-07:
Sorry for the delay, but I had a little issues with getting working (I fried my only IR led :) ). Today I bought new one and had success trying it on the table.

Here are some of my notes:
 - It doesn't seem to do any false triggers (not yet at least), but it usually double (or more) triggers when moving the IR receiver over the IR leds. See attached picture. Debug 3 corresponds to logic level of input pin on IR receiver.
- The IR receiver goes low when receiving pulses. It doesn't stay low for long, internal filtering probably kicks in, and start to filter out the signal and goes back to high after few ms. Then the filter probably drops of, and IR receiver goes low again. This repeats for a while until it settles, so until that happens, signal oscillates a lot.
- Because of that, we will need some sort of filtering.  I gonna add some time of insensitivity after signal goes back to high after trigger and see how that goes.
- Code in #1931 uses pull-down resistor in MCU. Manufacturer recomends 10Kohm pull up, so we probably should use pull-up instead.

[image]

jflyper, 2017-01-07:
Give me 0.5h.

DaveLister007, 2017-01-07:
Here is code for arduino used to generate 38Khz signal for transmitting led, if anybody wants to try it: 

```
#include <IRremote.h>

#define PIN_IR 3
#define PIN_DETECT 2
#define PIN_STATUS 13

IRsend irsend;
void setup()
{
  pinMode(PIN_DETECT, INPUT);
  pinMode(PIN_STATUS, OUTPUT);
  //pinMode(PIN_IR, OUTPUT);
  irsend.enableIROut(38);
  irsend.mark(0);
}

void loop() {
  //digitalWrite(PIN_STATUS, !digitalRead(PIN_DETECT));
  digitalWrite(PIN_STATUS, 1);
  delay(1000);
  digitalWrite(PIN_STATUS, 0);
  delay(1000);
  
}
```

jflyper, 2017-01-07:
Okay, simple guard time added.

closed 2022-05-09 by haslinghuis