betaflight/betaflight #1925: Blackbox - DTerm / Yaw filter are not displayed ...

TheAngularity, 2016-12-29:
... in the bf blackbox explorer :((
dterm lpf / yaw lpf and dterm notch filters are not displayed in blackbox explorer but they are correctly processed ... with gyro notch is all ok ... is this a firmware or a explorer issues

GaryKeeble, 2016-12-29:
Hi,
This is something related to the flight controller as the fields in the log file are actually zero; so the viewer is showing them correctly.

I have a very recent build for an Omnibus F3 and that logs those fields without issue; could you post a diff of your configuration; I know that the dterm lpf, yaw lpf and dterm notch filters are profile specific; so maybe its logging the wrong profile settings?

TheAngularity, 2016-12-29:
have the problem on both fc, airbotf4 and spracingf3evo ... profile has not been changed

here the diff matching the log above:
```
# Betaflight/SPRACINGF3EVO 3.1.0 Dec 29 2016 / 08:56:32 (4191afd)

# name
name TruX220

# resources

# mixer

# servo

# servo mix

# feature
feature -RX_PPM
feature -TELEMETRY
feature -CURRENT_METER
feature -RSSI_ADC
feature -TRANSPONDER
feature RX_SERIAL
feature LED_STRIP
feature AIRMODE

# beeper

# map
map TAER1234

# serial
serial 0 0 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# led
led 0 11,10::AW:0
led 1 10,10::AW:0
led 2 9,10::AW:0
led 3 8,10::AW:0
led 4 7,10::AW:0
led 5 6,10::AW:0
led 6 5,10::AW:0
led 7 4,10::AW:0
led 8 9,5::CW:6
led 9 8,5::CW:6
led 10 7,5::CW:6
led 11 6,5::CW:6
led 12 6,2::CW:0
led 13 7,2::CW:0
led 14 8,2::CW:0
led 15 9,2::CW:0

# color

# mode_color
mode_color 6 1 3

# aux
aux 0 0 0 1800 2000
aux 1 12 1 1000 1200
aux 2 14 1 1000 1200
aux 3 14 1 1800 2000

# adjrange

# rxrange

# rxfail

# master
set min_check = 1000
set max_aux_channels = 2
set max_throttle = 1950
set use_unsynced_pwm = ON
set motor_pwm_protocol = MULTISHOT
set motor_pwm_rate = 4000
set serialrx_provider = SUMD
set vbat_scale = 112
set align_board_yaw = 90
set gyro_notch1_hz = 370
set gyro_notch1_cutoff = 340
set gyro_notch2_hz = 170
set gyro_notch2_cutoff = 120
set acc_hardware = NONE
set pid_process_denom = 2
set blackbox_rate_denom = 4
set ledstrip_visual_beeper = ON

# profile
set dterm_notch_hz = 280
set dterm_notch_cutoff = 230
set pid_at_min_throttle = OFF
set p_pitch = 43
set i_pitch = 40
set d_pitch = 20

# rateprofile
set rc_rate = 108
set rc_rate_yaw = 108
set roll_srate = 80
set pitch_srate = 80

```

TheAngularity, 2016-12-30:
@GaryKeeble 
found it ... I'm pretty sure it's related to this #1865

TheAngularity, 2016-12-30:
here another screenshot ... there is absolutely something wrong with the blackbox:

[image]

GaryKeeble, 2016-12-30:
@borisbstyle, @TheAngularity 

The values in the log file are corrupt; those that reference these two new variables in blackbox.c: -

`const profile_t *currentProfile = &masterConfig.profile[masterConfig.current_profile_index];`

and

`const controlRateConfig_t *currentControlRateProfile = &currentProfile->controlRateProfile[masterConfig.profile[masterConfig.current_profile_index].activeRateProfile];
`

Could it be that something is wrong with how the pointers are assigned ?

martinbudden, 2016-12-30:
OK, I think I see the problem. Fix on the way.

martinbudden, 2016-12-30:
@TheAngularity , fixed submitted and merged. Can you try the latest build once it has completed?

TheAngularity, 2016-12-30:
bench test with airbotf4 successful ... all values match again 👍

martinbudden, 2016-12-30:
Great. Closing issue.

closed 2016-12-30 by martinbudden