betaflight/betaflight #1930: ESC Serial Passthrough broken when using motor outputs on same timer

CC-Jeff, 2016-12-30:
I'm using a C# application to talk to multiple ESCs connected through a flight controller, and have noticed that betaflight becomes unresponsive if I try to establish communication with an ESC whose output timer matches one already used. For example, on an SPRacingF3, I can talk to ESCs 1 (TIM16), 2 (TIM17), and 3 (TIM4), but if I try to talk to ESC 4 (also TIM4) the FC becomes unresponsive and needs to be rebooted. However, if I try to talk to ESC 7 (TIM15) instead, there's no conflict and everything works fine.

On an oscilloscope, I see the conflicting ESC signal changes when I write the passthrough command, but after that it stops responding. Any serial packets I send are ignored entirely, including the escape command (MSP type = 0xF4). It looks like I can get around it by always sending the reset command between escaping and establishing passthrough, but obviously that's not much of a solution.

I'm using MSP for establishing the communication, but I also saw this when using CLI. My Naze seems to have the same issue, but most of my testing has been on the SPRacingF3. I'll be taking a crack at fixing this tomorrow and after the holiday, but I'm still pretty unfamiliar with betaflight's code. If someone more experienced would like to take a crack at it I'd be happy to assist in testing.

mikeller, 2016-12-30:
I had a quick look at the escserial code, and there seems to be a discrepancy of what timer gets assigned for a particular ESC, and what timer gets deassigned:

https://github.com/betaflight/betaflight/blob/master/src/main/drivers/serial_escserial.c#L951-L956

vs.

https://github.com/betaflight/betaflight/blob/master/src/main/drivers/serial_escserial.c#L985

Replacing `output` in L985 with `motor_output` will potentially solve this. (Even though I think the code in L951ff is suspect and prone to breaking too.)

CC-Jeff, 2016-12-30:
> Replacing output in L985 with motor_output will potentially solve this. 

It did, worked like a charm. I also tried only passing the port index and removing the hardware reassignment entirely:
```
void closeEscSerial(escSerialPortIndex_e portIndex)
{
    escSerial_t *escSerial = &(escSerialPorts[portIndex]);

    escSerialInputPortDeConfig(escSerial->rxTimerHardware);
    timerChConfigCallbacks(escSerial->txTimerHardware,NULL,NULL);
    timerChConfigCallbacks(escSerial->rxTimerHardware,NULL,NULL);
    TIM_DeInit(escSerial->txTimerHardware->tim);
    TIM_DeInit(escSerial->rxTimerHardware->tim);
}
```
and that seems to work. No reason to potentially change the timer just before it's deassigned, right? Though I'm not sure what happens here if the mode is PROTOCOL_KISSALL since I don't think `rxTimerHardware` gets initialized, but it looks like previously it would try to set down timer `255` of `timerHardware[]`, which doesn't seem ideal. What do you think the correct behavior would be for this case? Set down `txTimerHardware` and leave `rxTimerHardware` alone? Or is it sufficient just to change the parameter in L985 as you suggested and leave the rest alone?

PR #1985 (2017-01-04): Fix CloseEscSerial to set down correct timer
Resolves #1930 
Works with Castle ESCs; I don't have KISS ESCs to test `PROTOCOL_KISSALL` with, but the signals appear correct on an oscilloscope when I send that command.
closed 2017-01-04 by mikeller