betaflight/betaflight #1938: BlackBox partly "wrong" values - BF 3.1 Build 861

SoulPictures, 2016-12-30:
hi there,

i just flashed 3.1 Build 861, now my BlackBox data looks like this:

[image]

PID_P, PID_I and PID_D look fine, but Gyros and Motors got extreme values.

Hardware used:

Motolab Tempest
BB via Serial (OpenLog with 16GB SanDisk Ultra @500Hz)
BB at UART1 with 250000
4/4/4kHz ( Gyro Only ) DShot300

As soon as i raise throttle to higher than 1500 i get useful Motor data

[image]

Gyro still looks strange although:

[image]

PID Error respectively looks strange too

[image]

borisbstyle, 2016-12-30:
Use the latest blackbox app from github

closed 2016-12-30 by borisbstyle

SoulPictures, 2016-12-30:
Thanks ! Works, will report remaining issues with BB there.