betaflight/betaflight #1941: BF 3.1 Airmode startup: no controls active when throttle is below xx threshold? (SPRacing F3 target)

RipperDrone, 2016-12-30:
New behavior on my F3's: When arming (in airmode), throttle stick triggers motors spinning uniformly, but below a certain throttle threshold, no roll/pitch/yaw axes control inputs are recognized. Only when I'm above some threshold (min_throttle?), then steering controls are active.

Can be quite surprising when you want to do flips / rolls in airmode with throttle @ zero, isn't it?

Behavior on 3.0.1 was that all inputs were active once throttle has been touched after arming.

borisbstyle, 2016-12-30:
Behaviour is exactly the same like on 3.0.1

Only if you have pid_at_min_throttle = OFF the behaviour should be like you describe

RipperDrone, 2016-12-30:
ok, then it's more of a useful feature than a bug. Will close this issue then. Thank you!

closed 2016-12-30 by RipperDrone