betaflight/betaflight #1942: RSSI scaling changed in BF 3.1? SPRacingF3 target

RipperDrone, 2016-12-30:
I just noticed that on y'day's Nightly Build, RSSI scaling seems to be in the wild.

I tried to change it via rssi_scaling, didn't match up with what it was before. I think it was a 100dB based scale before, wasn't it?

Right now, my Taranis shows RSSI = 1 in FrSky telemetry screen when I'm close. I bumped the Taranis scale factor up to 100, now RSSI is around 30 when I'm close and lower when I'm walking away.

Any hints on RSSI scaling changes or maybe different scaling of telemtry based RSSI only?

DieHertz, 2016-12-30:
RSSI is handled by your receiver, not BF. Maybe something changed in OpenTX?

RipperDrone, 2016-12-30:
Good point - progressed to OTX 2.2 to try out live PID adjustment from Taranis.

So RSSI is not passed on to FC and subject to RSSI_scale, correct? In this case we can close this issue, will have to look into OTX...

DieHertz, 2016-12-30:
RSSI seen on your Taranis is exclusively handle by your Tx and Rx, no FC involved

RipperDrone, 2016-12-30:
thanks, closed then. Will post in OTX 2.2 branch... Have a great New Year!

closed 2016-12-30 by RipperDrone