betaflight/betaflight #1954: diff command should not print unchanged sections

borisbstyle, 2016-12-31:
For example led, color and mode_color are unchanged. Therefore it should not be printed to keep the diff output more clean. What do you think @mikeller ?

```
# serial
serial 0 0 115200 57600 0 115200

# led

# color

# mode_color
```

MJ666, 2017-01-03:
Without this headings (even without all the headings for an diff) the output would be more clean and more easy to read.

mikeller, 2017-01-03:
@borisbstyle: Agreed that it would be an improvement to suppress output of empty headers. But currently the dump / diff is generated and sent line by line, so doing the checks if there was going to be output before every header would add a lot of complexity, which does not seem to be justified for the cleanup it achieves.

@MJ666 I guess the beauty lies in the eye of the beholder here. To me, the structured form with proper headings is way easier to understand than just a bunch of `set` commands without context. Also, for some things (like profile / rateprofile switching), the comments provide some context as to why things are done.

borisbstyle, 2017-01-03:
@mikeller 
yes I know that. I figured out that after I created the issue. You would have to first perform compare of each header section if there are any changes prior to printing

mikeller, 2017-01-03:
Yes. Probably something we can look into once we've switched to parameter groups, if we rearrange the headers to match the groups, and then do a binary compare of the config.

closed 2022-05-09 by haslinghuis